MIPT-ILab/mipt-mips #17: Implement Data Cache Unit and introduce it into the pipeline (4 points)

pavelkryukov, 2016-04-14:


pavelkryukov, 2018-02-27:
On cache hit, pipeline operates as usual
On cache miss, pipeline is stalled. Address of missed line is pushed to a long-latency port which is monitored by separate method (`clock_cache`). When data is returned, it is filled into cache so pipeline is unstalled.

pavelkryukov, 2018-05-11:
@trexxet I assigned you to severing tasks that you may complete during summer. They are 20 points, covering all requirements for 1st and 2nd term.

pavelkryukov, 2018-06-15:
@trexxet Do you read my e-mails?

trexxet, 2018-06-15:
@pavelkryukov 
I apologize for the delay. I didn't check e-mail due to exams.
I'll take on these tasks right after exams (approximately 22-25 Jun), and I'm sure I'll complete them by the end of summer. However, I need to examine each problem to estimate time to complete them, so I'd like to do it after exams. I hope this wouldn't delay any work on project.

pavelkryukov, 2018-06-25:
@trexxet just a friendly reminder, it is June 25 today...

pavelkryukov, 2018-07-01:
Ok.

> I think I'll begin with #290
Not sure it is a very simple thing, as it requires some deep dive into GDB internals.

What do you mean by #18? It is completed already IMO.

trexxet, 2018-07-01:
As I see, for #18 some research should be finished to explain logarithmic law.