MIPT-ILab/mipt-mips #18: Implement Instruction Cache Unit and introduce it into pipeline (4 points)

pavelkryukov, 2016-04-14:


pavelkryukov, 2018-02-27:
On cache hit, proceed as is.
On cache miss, push bubble into pipeline and send a request to long-latency loop port to model memory access. Special thread listens to that port and fill cache with actual data.

pavelkryukov, 2018-02-27:
@alex19999 — I think you almost completed the pipeline part of the task. Would you like to proceed with introduction of InstrCache class?

pavelkryukov, 2018-02-27:
Good!

You may find detailed description of CacheTagArray here: https://github.com/MIPT-ILab/mipt-mips/wiki/Cache-model

Default values for parameters should be:
* size = 2048
* ways = 4
* line size = 64

alex19999, 2018-02-27:
Ok , thank you, I will explore this information and may be ask you some questions later.27.02.2018, 19:32, "Pavel I.

pavelkryukov, 2018-02-28:
There is no dedicated trace to test IC, but you may use `dc_ic_stress.s`. By chaning cache parameters, you should see performance results close to the chart from L11. 

See this page to get list of all cache configurations and fill the IPC numbers: https://github.com/MIPT-ILab/mipt-mips/wiki/cache-associativiy-studies

pavelkryukov, 2018-03-13:
Not closing that as there are next steps:
* [ ] Cache size, associativity, and line size should be configurable with Config objects
* [ ] Sensitivity to cache line size should be demonstrated with `dc_ic_stress.s`. If cache line is 4 bytes, each instruction is a cache miss → AMAT=30. If cache line is 8 bytes, each second instruction is a cache miss → AMAT=15. Since `dc_ic_stress.s` does not contain loops, there is no much sensitivity to cache size and/or # of ways.

alex19999, 2018-03-13:
Pavel Igorevich, may I change 
`create_simulator( const std::string& isa, bool functional_only, bool log)` function?( I mean I am going to add cache parameters(`uint32 size_in_bytes`, `uint32 ways`, `uint32 line_size`))

pavelkryukov, 2018-03-13:
It's better to put these configuration objects into `fetch.cpp`

alex19999, 2018-03-13:
Allright, thank you

alex19999, 2018-03-13:
And about task 2: As I've understood, the task is to make cache cache associativity studies. What files should be in the commit?

pavelkryukov, 2018-03-13:
> What files should be in the commit?

My original intention is to have a Wiki page with chart, similar to what we got from standalone cache. But:

> Since dc_ic_stress.s does not contain loops, there is no much sensitivity to cache size and/or # of ways.

I realized that the only IC hungry trace `dc_ic_stress.s` is not a trace to run such study. It is just a simple stream of instructions, so all cache misses are 'cold' misses. Moreover, if you add a backward loop from the end of trace to the beginning, you won't see results neither since instructions from the end of trace replaced ones from the beginning of trace.

So the only effective parameter is cache line size which corresponds to throughput of our fictionuous memory->cache interface. You may just put results with different options here.

alex19999, 2018-03-14:
So, as I understood, I only need to count the AMAT depending on the size of the cache line and leave these statistics here using `dc_ic_stress.c`. This is not very important, but how many instructions should I take? Сorrect me if I was wrong somewhere, please.

pavelkryukov, 2018-03-14:
Use may skip AMAT and use IPC instead. I would use 400.000 instructions.

alex19999, 2018-03-14:
Constant parameters:

- Instruction cache size in bytes = 2048;

- Amount of ways in instruction cache = 4;

- `dc_is_stress.s` was used for study with 400.000 instructions;

### Cache study results
Line size of instruction cache (in bytes) | IPC
------------ | -------------
4 | 0.041842
8 | 0.0803065
16 | 0.148593
32 | 0.25847
64 | 0.410101
128 | 0.580328
256 | 0.732299
512 | 0.842629

pavelkryukov, 2018-03-14:
The data on plot. Looks reasonable!

[image]

Thank you for your contribution!

closed 2018-03-14 by pavelkryukov

pavelkryukov, 2018-04-02:
@alex19999 

Could you please put you results into new Wiki page?

pavelkryukov, 2018-04-03:
Great. The page is here: https://github.com/MIPT-ILab/mipt-mips/wiki/IPC-sensitivity-to-cache-line-size

Please use structure I defined in the page. In general, it should be similar to reports you do in General Physics laboratory of MIPT.

alex19999, 2018-04-06:
Pavel Igorevich, I have finished this page. Could you review my work please?

pavelkryukov, 2018-04-07:
Thanks, looks good. I'll make grammar fixes later if needed.

pavelkryukov, 2018-04-07:
Please add some quantitative analysis which explains the logarithmic law.
There should be a simple formula to get IPC from cache line size and pipeline depth.

alex19999, 2018-04-07:
Pavel Igorevich, sorry, but it is important for this task to know percent of memory-access instructions, isn't it?
```
CPI = CPI(execution) + memory_stalls_per_instruction;
memory_stalls_per_instruction = memory_accesses_per_instruction * miss_rate * miss_penalty;
IPC = 1 / CPI;

```
and miss_rate is correlated with cache_line_size;

pavelkryukov, 2018-04-07:
Why? We study instruction cache, and each instruction is fetched from it...

alex19999, 2018-04-07:
Oh, yes, I forgot about it. Just think  about cache in general

commit 88c54593 (2018-03-13): Introduce Instruction cache unit into pipeline (fixes #18)