MIPT-ILab/mipt-mips #20: Implement non-unified pipeline latency (3 points)

pavelkryukov, 2016-04-14:


pavelkryukov, 2018-03-06:
Proper latency is a subject of microarchitecture. For now, let's consider following:

Control flow: execution (generation of condition) + memory (generation of result)
* beq bne blez bgtz beql bnel blezl bgtzl btlz bgez bltzl bgezl bltzal bgezal bltzall bgezall
* j jal jalx jr jalr
* movci movn movz
* syscall break tge tgeu tlt tltu teq tne tgei tlti tltiu teqi tnei

ALU-only: execution
* add addu sub subu and or xor nor dadd daddu dsub dsubu
* addi addiu slti sltiu andi ori xori lui daddi daddiu
* sll srl sra srlv srav slt sltu sllv dsllv dsrlv dsrav
* movz movn syscall
* mfhi mthi mflo mfh

Multiplication/Division: 3 stages of execution
* mult multu div divu dmult dmultu ddiv ddivu

Store/Load: execution + memory:
 * ldl ldr lb lh lwl lw lbu lhu lwr ll lwu lwc1 lwc2 pref lld ldc1 ldc2 ld
 * sb sh swl sw sdl sdr swr sc swc1 swc2 scd sdc1 sdc2 sd

denislos, 2018-03-06:
I think I should start here with the split of remained pipeline stages.

pavelkryukov, 2018-03-06:
Could you please explain?

denislos, 2018-03-06:
It would be nice to have decode, execute and memory stages in the separate classes.

pavelkryukov, 2018-03-30:
Do you have anything in progress? I would like to reorganize directories in repository a little, I don't want to interfere with your changes

denislos, 2018-03-30:
Yes, I have implemented an infrastructure of the complex pipeline. I will endeavour to describe my current progress:

- I have added new substages to Execute stage (class) to simulate ALU for complex arithmetic which is internally pipelined. They share an access to datapath ports (DECODE_2_EXECUTE, EXECUTE_2_MEMORY, EXECUTE_2_WRITEBACK) which are used in order to communicate with the world outside. Intersubstage datapath ports are used when an access to the complex arithmetic unit is required.
-  I have introduced a concept of the pipeline route to DataBypass class in order to update current stage of the traced instructions. I have also added `cycles_till_writeback` to RegisterInfo.
- Furthermore, taking into account Hennessy& Patterson, I made a questionable assumption on the way Writeback stage would be simulated. An idea was to have a writeback stage with a single writeport  to the register file but with multiple routes  so as to get several instructions completed at the same cycle in the writeback stage when all conditions for performed checks (these checks are performed in the `is_stall` method of DataBypass class) on RAW, WAW dependencies and structural hazards are satisfied. 

However, I faced some difficulties with the functional simulator. I have not come up with the nice solution to this problem yet except for some bad and obvious ones but I am looking forward to work it out as soon as possible. I am sorry for such a long delay.

I think I won't have any problems with integration of changes in repository structure.

pavelkryukov, 2018-03-30:
> Intersubstage datapath ports are used when an access to the complex arithmetic unit is required.

Is it possible to use data ports with increased latency instead of 'intersubstage' ports? They would allow to change latency dynamically, simulating different CPUs.

> An idea was to have a writeback stage with a single writeport to the register file but with multiple routes so as to get several instructions completed at the same cycle in the writeback stage when all conditions for performed checks

I think I'm missing something. If instructions complete at the same cycle, there cannot be a single write port to register file, right?

If you have some code that can be checked-in independently from your other changes, it will be nice to have it committed to main repository to ensure its safety from contribution of other developers.

denislos, 2018-03-31:
> Is it possible to use data ports with increased latency instead of 'intersubstage' ports? They would allow to change latency dynamically, simulating different CPUs.

Yes, but discarding invalid instructions ( and also bypassed data in some possible implementations) with a flush signal would become non-trivial. In my opinion, it would also lead to the reduced information value of disassembly. I will try to come up with an idea how mispedictions can be tackled in a nice way here.

> I think I'm missing something. If instructions complete at the same cycle, there cannot be a single write port to register file, right?

For example, I think `store` and `add` can approach the Writeback stage at the same cycle despite the  fact that `store` writes to zero register in our simulator.

pavelkryukov, 2018-03-31:
> For example, I think store and add can approach the Writeback stage at the same cycle

It is completely wrong. Please recall the latest lecture on memory disambiguation. In real HW stores write data _after_ retirement. We do it on mem stage just because in-order execution is guaranteed, so no younger load can read old data, and no older load reads updated data.

Additionally, store has a lot of opportunities to trap: unaligned address, segmentation fault, page permission fault, etc. Traps should be applied in-order, otherwise you may update the architectural state by next-after-trap instruction as well.

There are only two solutions:
1. Add a ROB to get real out-of-order engine (but not superscalar) – that is an over-engineering and dramatically increases complexity of maintaining
2. Keep in-order execution with scoreboarding.

pavelkryukov, 2018-03-31:
> Yes, but discarding invalid instructions ( and also bypassed data in some possible implementations) 

I do not see a problem here, there are two opportunities:
1. Make flush cycle active for N cycles, where N is the latency of port it affects
2. Add a flushing interface to the writing port (worse as it complicates code)

> In my opinion, it would also lead to the reduced information value of disassembly.

How?

denislos, 2018-03-31:
> It is completely wrong. Please recall the latest lecture on memory disambiguation. In real HW stores write data after retirement. We do it on mem stage just because in-order execution is guaranteed, so no younger load can read old data, and no older load reads updated data.

Thank you, I have realized the dullness of my proposal.

> How?
There would not be the information about the specific execute unit ( execute-0, execute-1, execute-2, etc.) which operates on a complex arithmetic instruction at the current moment.

pavelkryukov, 2018-04-17:
@denislos What is estimated time of arrival for the next pull request? I'm not urge you, I just want to know.

denislos, 2018-04-17:
I am sorry, I will open a pull request by the end of the week.

pavelkryukov, 2018-04-25:
Please proceed with documentation.

pavelkryukov, 2018-04-25:
The link to the page https://github.com/MIPT-ILab/mipt-mips/wiki/Data-Bypass-and-Scoreboard.
Please use that page as example: https://github.com/MIPT-ILab/mipt-mips/wiki/BPU-model

denislos, 2018-04-25:
Thank you, I will endeavour to complete it as soon as possible.

commit fa48c005 (2018-04-25): Implement complex pipeline(fixes #20) (#404)
closed 2018-05-04 by pavelkryukov