MIPT-ILab/mipt-mips #26: Use Boost for parameters parsing (1 point)

pavelkryukov, 2016-04-14:


pavelkryukov, 2017-02-17:
Boost is a set of C++ libraries used everywhere. One of the libraries is Program Options which allows to add and remove command line options easily. Documentation is avalilable here: http://www.boost.org/doc/libs/1_58_0/doc/html/program_options.html.

Years ago I did the same task for MDSP project, so you may re-use that work: https://github.com/MIPT-ILab/MDSP/blob/master/sources/funcsim/config.cpp

pavelkryukov, 2017-02-27:
@ivan23kor Any progress?

pavelkryukov, 2017-04-05:
Raising points to 3 as that issue is extremely important to go forward

gkorepanov, 2017-04-05:
Why is it so important? 
It means that the present arguments (filenames and sections, instructions number to be executed and so on) should be just parsed with boost?

pavelkryukov, 2017-04-06:
We are going to increase amount of parameters dramatically: cache sizes/associativity, instruction latencies, BP modes, levels of debugging information.

closed 2017-04-08 by pavelkryukov