MIPT-ILab/mipt-mips #28: Move GoogleTest library out of repository and add installation instruction (1 point)

pavelkryukov, 2016-04-15:


gkorepanov, 2017-03-08:
I wanted to start working on that issue, so I have read through GoogleTest lib docs. They suggest not to install a precompiled system-wide version of their lib, and the reasoning seems to be sane:
[Reference to Google Test docs](https://github.com/google/googletest/blob/master/googletest/docs/FAQ.md#why-is-it-not-recommended-to-install-a-pre-compiled-copy-of-google-test-for-example-into-usrlocal)

Do we really need to exclude it then?

pavelkryukov, 2017-03-08:
@gkorepanov 

I suggest you to start with tasks on bumping GoogleTest version.

> Why is it not recommended to install a pre-compiled copy of Google Test (for example, into /usr/local)?

I suppose we won't keep pre-compiled copy of Google Test on the machine, but it would be just checked out from separate Git repository. Anyway, let me think a little bit more about that — meanwhile you can update Google Test in repository to the latest version and get extra points.

pavelkryukov, 2017-03-10:
> They suggest not to install a precompiled system-wide version of their lib, and the reasoning seems to be sane

However, I found that we use precompiled version in our repository. Should that be fixed?

gkorepanov, 2017-03-18:
Probably `make test` should just execute something like `wget` or `git submodule` to make a local copy of `gtest` repo inside ours and then just compile it and check if everything is OK.

pavelkryukov, 2017-03-18:
@gkorepanov: Sounds good.

By the way, @alexander-titov told me that he had some issues on library compilation, that's why it is pre-built in repository. Probably we can rid of it with newer version.

gkorepanov, 2017-03-18:
I'm currently working on this, and seems everything is just fine with new version. I will remove the googletest source code from our repo completely, and it will be downloaded as one of the dependencies in Makefiles.

pavelkryukov, 2017-03-20:
Merged with #24, so there are 3 points now.

pavelkryukov, 2017-03-25:
@gkorepanov Any progress?

gkorepanov, 2017-03-25:
Yeah, I've completed what I've intended to do. However I have forgotten my laptop charger in Sochi and thus don't have an access to my files. Now I'm just reproducing what have already done.

pavelkryukov, 2017-03-25:
Done.

pavelkryukov, 2017-03-25:
Something is wrong with includes:

{{{
~/mipt-mips/func_sim/func_memory$ make testg++ -std=c++11 -c func_memory.cpp -I ./ -I ../..//common/ -I ../..//func_sim/elf_parser/
g++ -std=c++11 -c ../..//func_sim/elf_parser/elf_parser.cpp -I ./ -I ../..//common/ -I ../..//func_sim/elf_parser/
g++ -std=c++11 -I ../..//libs/googletest/lib/include -I ./ -I ../..//common/ -I ../..//func_sim/elf_parser/ -c unit_test.cpp
unit_test.cpp:6:25: fatal error: gtest/gtest.h: No such file or directory
compilation terminated.
Makefile:62: recipe for target 'unit_test.o' failed
make: *** [unit_test.o] Error 1
}}}

gkorepanov, 2017-03-26:
Thank you very much for mentioning the problem. I have forgotten to add proper deps to the rest of Makefiles.

closed 2017-03-26 by pavelkryukov