MIPT-ILab/mipt-mips #31: Implement data bypassing

pavelkryukov, 2016-04-18:


pavelkryukov, 2018-02-21:
@denislos Do you have anything to share or discuss?

denislos, 2018-02-24:
I have got 0.88 IPC ( 2832 cycles instead of 4762 for 2503 instructions) on tt.core.
There is a list of problems and my solutions to them:
1) conditional moves
Solution: I have added is_conditional_move() to mips_instr to check whether an instruction is conditional move or not. They are not bypassible.
2) HI/LO
Solution: It is possible to bypass from `div/mul` and `mthi/mtlo` because I have added special flags to scoreboard in order to check whether HI and LO registers are being processed by `div/mul` instructions or not. So the consumers receive the information whether they should get HI from bypassed data or carry it as an ordinary one.
3) mispredictions
Solution: I have implemented cancel( const MIPSInstr& instr) method to handle them, it's quite similar to the one we had in mips_rf.
4) I cope with WAW(#45) keeping the cycle of DECODE stage( when an instruction will be moved to EXECUTE) of an instruction.

There are still improvements to be made.

pavelkryukov, 2018-02-24:
> Solution: I have added is_conditional_move() to mips_instr to check whether an instruction is conditional move or not. They are not bypassible.

You may add `is_bypassable()` in addition to `is_conditional_move()`, so other MIPS instructions could be added as well.

> Solution: It is possible to bypass from div/mul and mthi/mtlo because I have added special flags to scoreboard in order to check whether HI and LO registers are being processed by div/mul [...]

Sounds good. I'm not really sure that these registers are available through bypasses in real HW, but let's keep it — I'll check it in the MIPSfpga later.

> Solution: I have implemented cancel( const MIPSInstr& instr) method to handle them, it's quite similar to the one we had in mips_rf.

OK. So MIPS RF and you class have almost the same interfaces, right?

> I cope with WAW(#45) keeping the cycle of DECODE stage( when an instruction will be moved to EXECUTE) of an instruction.

That practice is legal, but it slightly decreases maintainability of code, according to my experience. Keep it for now, I'll take a look at review.

pavelkryukov, 2018-02-24:
> Solution: It is possible to bypass from div/mul and mthi/mtlo because I have added special flags to scoreboard in order to check whether HI and LO registers are being processed by div/mul [...]

Do I understand correctly that $t4 will be equal to $t3 if following code is executed:
```
div $t0, $t1
mthi $t3
mfhi $t4
```

denislos, 2018-02-24:
My class has
bool `is_in_RF( RegNum num)`
bool `is_bypassible( RegNum num)`
BypassCommand `get_bypass_command( RegNum num)` - here i get the information if i should get data from "execute-execute" or "memory-execute" ports and if I should get HI (received_data >> 32) or Regular (received_data) from the port. It will be called on the decode stage and then its result will moved to EXECUTE stage using a specific port to give an idea what should be done there, if is_bypassible returns true. 
void `update( instr, cycle, next_stage_of_destination_register)` - to update scoreboard
void `cancel( instr)` - to cope with flushes.

Yes, $t4 will be equal to $t3.

denislos, 2018-02-24:
I will try to describe the way it works.
I have added to simulator these ports.
`std::unique_ptr<WritePort<uint64>> wp_execute_2_execute_forward = nullptr;
 std::unique_ptr<ReadPort<uint64>> rp_execute_2_execute_src1_forward = nullptr;
 std::unique_ptr<ReadPort<uint64>> rp_execute_2_execute_src2_forward = nullptr;

 std::unique_ptr<WritePort<uint64>> wp_memory_2_execute_forward = nullptr;
 std::unique_ptr<ReadPort<uint64>> rp_memory_2_execute_src1_forward = nullptr;
 std::unique_ptr<ReadPort<uint64>> rp_memory_2_execute_src2_forward = nullptr;

 std::unique_ptr<WritePort<DataBypass::BypassCommand>> wp_decode_2_execute_src1_command = nullptr;
std::unique_ptr<ReadPort<DataBypass::BypassCommand>> rp_decode_2_execute_src1_command = nullptr;

std::unique_ptr<WritePort<DataBypass::BypassCommand>> wp_decode_2_execute_src2_command = nullptr;
std::unique_ptr<ReadPort<DataBypass::BypassCommand>> rp_decode_2_execute_src2_command = nullptr;`

I have decided to let producers( EXECUTE and MEMORY stages) put data into "execute-execute" and "memory-memory" ports freely in order to have implementation close to HW where multiplexers are responsible for selection. In my implementation I get an information if these ports should be ignored or not from "command ports". BypassCommand consists of a direction(exe -> exe, mem->exe) and a specifier( Regular or HI (data>>32)).

pavelkryukov, 2018-02-24:
Please push a code to your fork to let me understand better

denislos, 2018-02-24:
Done

pavelkryukov, 2018-02-24:
Please check my comments.

You may submit a pull request to get everything under CI testing and proceed with fixes after that.

pavelkryukov, 2018-02-26:
I merged your pull request as it is a stable version, but please pay attention to my comments and do required changes.

closed 2018-03-01 by pavelkryukov

pavelkryukov, 2018-03-30:
Hi Denis

I'm thinking about overall simulator architecture, and I have an idea to let instruction have two destinations, for MIPS and RISC-V multiplication. It would simplify register file much, but I would like to check whether it does not explode data bypasses. Currently you are able to push 1 register to 2 sources of 3 stages, so there are 6 routes; with 2x destination we end up with 12 routes.

One benefit is that you can easily disable bypass of HI register then (I believe it is not a thing in actual HW as it is a lot of wires handling quite rare case). Alternatively, we may remove its simulation completely until someone demands it.

denislos, 2018-04-01:
I am sorry for not a quick response. In my opinion,  bypassing would be fine as soon as we were able to proceed with the right destination with the help of BypassCommand. I have also recently simplified DataBypass class by removal of flush notifiers because we won't need them until we want to have pipelined memory or writeback stages.
    I think that bypass of HI register should be removed because its benefit does not cover the cost of code complexity. Should I examine the case more thoroughly in order to have some working prototypes with two different destination fields?

pavelkryukov, 2018-04-01:
> I think that bypass of HI register should be removed because its benefit does not cover the cost of code complexity. Should I examine the case more thoroughly in order to have some working prototypes with two different destination fields?

As far as I see, integration of 64 bit instructions is highly dependent on existing 2x routing. Let's wait for finalization of works here and see.