MIPT-ILab/mipt-mips #35: Fork mem_trans.s to generate ICU stress test

pavelkryukov, 2017-02-02:
Currently mem_trans.s is designed to test DCU, but we might want to test ICU. So, the trace should contain non-memory instructions.

pavelkryukov, 2017-02-17:
@ivan23kor 

That is very simple task although important one. You have to replace all memory instructions to non-memory instructions in copy of mem_trans.h.
MIPS instructions https://github.com/MIPT-ILab/mipt-mips/wiki/MIPS-Instruction-Set. However, our simulator does not support all of them, here is the list of all supported instructions: https://github.com/MIPT-ILab/mipt-mips/blob/master/func_sim/func_instr/func_instr.cpp

pavelkryukov, 2017-02-27:
@ivan23kor Please share your progress on this.

ivan23kor, 2017-02-28:
Sorry to tell you, I have a lack of time so I won't commit to the
repository.

gkorepanov, 2017-03-08:
I'm sorry for a couple of silly questions:
1. I'm wondering what are the DCU and ICU?
2. `find` haven't gave me any file named `mem_trans.s` in our repo. Am I missing something?

igorsmir-ilab, 2017-03-08:
Hello,

1. **DCU** stands for "Data Cache Unit", while **ICU** means "Instruction Cache Unit". They are also called **L1D** and **L1I**, where L1 means "level 1" cache. You may find processor terminology e.g. in section 1.2 of Intel [cheatsheet](http://www.intel.com/content/dam/www/public/us/en/documents/datasheets/xeon-e7-v2-datasheet-vol-2.pdf).
2. I believe @pavelkryukov meant our memory test, which is currently located under `tests/samples` and called `memtrace_based.s`, not `mem_trans.s`.

pavelkryukov, 2017-03-08:
Yep, sure I meant memtrace_based.s

gkorepanov, 2017-03-09:
I'm really grateful for your explanations and references. I believe it will help me to proceed.

pavelkryukov, 2017-03-22:
@gkorepanov Any progress?

pavelkryukov, 2017-06-22:
Kamino closed and cloned this issue to MIPT-ILab/mips-traces