MIPT-ILab/mipt-mips #36: For mem_trans.h to test BP

pavelkryukov, 2017-02-02:
Currently no testing for Branch Prediction is made. We have to add the trace with huge amount of random branches.

**AMB:** Trace is implemented and shows sensitivity to BP size.

gkorepanov, 2017-04-10:
How do I check that the trace shows the sensitivity to BP size, if there is no BPU yet?

gkorepanov, 2017-04-10:
Could you please construe how the `memtrace_based.s` was created? Is there any generator?
And what do you mean by `random branches`? If they are truly random, the branch prediction won't work at all.

I'd be grateful for explanations.

pavelkryukov, 2017-04-10:
I think @igorsmir-ilab will help you on this,

>Could you please construe how the memtrace_based.s was created? Is there any generator?

@igorsmir-ilab Do you remember? It was generated from mem_trace.txt.

> And what do you mean by random branches? If they are truly random, the branch prediction won't work at all.

The goal is to make a kind of stress test: implement trace with many branches (so they won't fit to small BPU) and different behavior. I suppose such traces exist already, you can try to find them.

> How do I check that the trace shows the sensitivity to BP size, if there is no BPU yet?

So you demonstrate it when BP is implemented.

igorsmir-ilab, 2017-04-10:
Yes, this trace was created from `mem_trace.txt`, which just contained different mem addresses used for cache simulation task. This trace divided into two parts, as I remember: initialization part with `store` instructions, and memory accesses themselves (lots of `load` instructions). Addresses values and their order in this trace should be similar to original `mem_trace.txt`
There was also a small trick to decrease the end size of `memtrace_based.s`: I've detected the long chains of sequential memory accesses and squashed them in a single loop, rather than simply put the whole list of `loads`/`stores`.

igorsmir-ilab, 2017-04-10:
I'm not sure `memtrace_based.s` is the best trace for BPU stress test base, since there are multiple unique loops without inner branches. Like @pavelkryukov said, you may look around and check, whether do we already have some nice tests for BPU or not. Nevertheless, building synthetic test with multiple branches should be quite easy, IMO.

pavelkryukov, 2017-04-24:
Moved to the next year

pavelkryukov, 2017-06-22:
Kamino closed and cloned this issue to MIPT-ILab/mips-traces