MIPT-ILab/mipt-mips #37: Enable PVS-Studio in MIPT-MIPS

pavelkryukov, 2017-02-11:
Recently PVS-Studio developers provided a free service to use their static analyzer tool. We would like to use in our project.

Information: http://www.viva64.com/ru/b/0457/

AMB: PVS Studio support enabled in MIPT-MIPS

pavelkryukov, 2017-02-18:
@MrLightningRod Does it require MSVS build?

gkorepanov, 2017-03-08:
Am I right thinking that our project should be positioned as 

`an open source non-commercial project` 

rather than

`personal academic project`?

gkorepanov, 2017-03-08:
And where should I post the results of static analysis?

pavelkryukov, 2017-03-08:
1. «This is an open source non-commercial project»
2. I think the best way is to open a new GitHub tracker and enumerate all findings here.

pavelkryukov, 2017-03-22:
@gkorepanov Any progress?

pavelkryukov, 2017-04-03:
@gkorepanov Any updates?

pavelkryukov, 2017-10-17:
@alex19999 I see you have added the comments. Have you obtained feedback from PVS-studio?

pavelkryukov, 2017-10-19:
@alex19999 Why do you ignore my questions? :-)

pavelkryukov, 2017-10-22:
@alex19999 Could you please share the command line you used? I will add it to our Travis build.

pavelkryukov, 2017-10-22:
Yes. We need a set of bash commands which one should run to get analysis results.

alex19999, 2017-10-22:
Ok, I used sush comandspvs-studio-analyzer trace -- makepvs-studio-analyzer analyze -l ~/.config/PVS-Studio/PVS-Studio.lic  -o ./project.log   -j8 --compiler COMPILER_NAME --compiler gcc --compiler g++ --compiler COMPILER_NAMEplog-converter -a GA:1,2 -t errorfile ./project.err ./project.log22.10.2017, 12:05, "Pavel I.

PR #205 (2017-10-23): Enable PVS-Studio
Fixed #37 , using right codestyle
closed 2017-10-23 by pavelkryukov