MIPT-ILab/mipt-mips #44: Enable very strict C++ warning rules

pavelkryukov, 2017-03-10:
I personally prefer to enable as much C++ warnings as it is possible with `-Wall -Wextra -Werror` flags. Can someone please enable these flag and fix warnings appeared?

**AMB:** These flags are enabled in all Makefiles, all builds pass.

closed 2017-04-05 by pavelkryukov

gkorepanov, 2017-04-05:
After enabling all the strict rules I can not build a single program, every one has some warnings (interpreted as errors). Should I try to fix them?

(I wonder how that is possible, I see that you have accomplished a great amount of work within the repository to enable strict flags and newer C++)

pavelkryukov, 2017-04-06:
Please send a command line and output

gkorepanov, 2017-04-06:
### func_sim
```
$ make 
g++ -std=c++14 -Wall -Wextra -Werror -Wno-sign-compare -O3 -c ..//func_sim/func_memory/func_memory.cpp -I ..//
In file included from ..//func_sim/func_memory/func_memory.cpp:16:
..//func_sim/func_memory/func_memory.h:27:16: error: private field 'addr_bits' is not used
      [-Werror,-Wunused-private-field]
        uint64 addr_bits;
               ^
```

gkorepanov, 2017-04-06:
### elf_parser

```
$ make
g++ -std=c++14 -Wall -Wextra -Werror -Wno-sign-compare -O3 -c elf_parser.cpp -I ../../
In file included from elf_parser.cpp:24:
./elf_parser.h:14:1: error: class 'Elf' was previously declared as a struct
      [-Werror,-Wmismatched-tags]
class Elf;
^
/usr/local/include/libelf/libelf.h:128:16: note: previous use is here
typedef struct Elf      Elf;
               ^
```

pavelkryukov, 2017-04-06:
And what is g++ version?

gkorepanov, 2017-04-06:
Wow. That was the clue. I've recently switched from Linux to MacOS and it was a surprise for me that in MacOS the gcc is just linked to clang. I believe everything will work when I run real gcc.

pavelkryukov, 2017-04-06:
Yep, clang has these annoying warnings about 'struct/class' mismatch

gkorepanov, 2017-04-06:
Everything works just fine with gcc 6.3.0. Thanks for help.