MIPT-ILab/mipt-mips #48: Add root Makefile to keep all compiler/linker options

pavelkryukov, 2017-03-15:
Currently we define CXXFLAGS in each Makefile. The object is to add one Makefile in $TRUNK to keep all the options.

**AMB:** root Makefile is added, all other Makefiles include it,

gkorepanov, 2017-03-25:
Well, I have studied the question a bit. I'm wondering what exactly we are going to achieve. The variants are:

- The **recursive build system** which allows to build any module (like `func_instr`) independently or to build a whole project.
This is mainly considered not to be the best way to go: [Recursive Make Considered Harmful](http://aegis.sourceforge.net/auug97.pdf)

- The **non-recursive build system** allowing for different targets and wise multi-level include system, which would allow to build the whole project in a moment as well (I do consider this way to be best, as the project grows and we are already having cross-dependencies (just look at the `perf_sim` Makefile with a number of `vpath` statements, or probably consider the `gtest` lib necessary for almost any module))

- Just **few independent Makefiles** sharing the common code in the top-level one (it's the closest to what we have now but don't seem to be brilliant in my view)

- Whatever else...?

pavelkryukov, 2017-03-25:
@igorsmir-ilab Igor, what do you think?

igorsmir-ilab, 2017-03-28:
I would say let's stick to the easiest option for now - **few independent Makefiles** with shared root. This will allow us to solve current task - central CXXFLAGS definition, as well as enable faster forward progress with very strict C++ warnings issue [#44](https://github.com/MIPT-ILab/mipt-mips/issues/44).

The **non-recursive build system** could be set as the next milestone, IMO.

gkorepanov, 2017-04-05:
I'm now adding the header dependencies to all Makefiles and it looks that all of them (Makefiles) are becoming quite similar. Probably I should make a root Makefile with common rules as well?

pavelkryukov, 2017-04-05:
It is not critical now but can create some problems to current users as they won't expect significant build changes. Let's postpone that for now.

gkorepanov, 2017-04-05:
Ok, thanks.

pavelkryukov, 2017-04-05:
WIP on testing changes

closed 2017-04-05 by pavelkryukov