MIPT-ILab/mipt-mips #69: Use faster C++11 STL container for MIPS ISA

pavelkryukov, 2017-04-10:
Currently MIPS ISA is stored in C-style static array. With C++11 we may initialize something like std::list or std::unordered_map during compile-time.

pavelkryukov, 2017-10-15:
@TimofeevAlex I would like to see 3 `std::unordered_maps`. If opcode is 0x0, search is performed over map of R-instructions. If opcode is 0x1, search is performed over map of RI-instructions (so it is a good opportunity to close issue #93 and get 3 more points!). Otherwise, search is performed over map of I- and J-instructions.

pavelkryukov, 2017-10-23:
@TimofeevAlex Any progress?

TimofeevAlex, 2017-10-23:
I haven't had enough time to do this task yet. I've planned to do it
seriously next week.

pavelkryukov, 2017-10-24:
@TimofeevAlex Ok, thank you for notification. I advise you to finish it next week as we have a MIPS ISA lecture in the beginning of November, so other guys may implement missing MIPS instructions, creating merge conflicts for you.

TimofeevAlex, 2017-11-01:
I have figured out the code in the project,  C++ container and started
writing the code. But I am afraid that I won't get in time.

pavelkryukov, 2017-11-01:
That's great!

I think we should use std::unordered_map here as it has O(1) read time.
Currently we have this:

```c++
    for ( const auto& entry : isaTable)
    {
        bool is_i_R = ( entry.format == FORMAT_R);
        if ( entry.opcode == ident && ( is_i_R == is_R))
        {
              // instruction initialization
```

And we want to get this (at least, you may do something better):
```с++
    const auto& entry = isaTable.at( std::make_pair( opcode, is_R));
    if ( entry != isaTable.end())
    {
         // instruction initialization
```

pavelkryukov, 2017-11-01:
Feel free to ask questions

TimofeevAlex, 2017-11-01:
OK I understood

pavelkryukov, 2017-11-08:
One week passed since our sync-up. Do you have any updates on the issue?

pavelkryukov, 2017-11-16:
One more week passed since our previous sync-up. Do you have any updates on the issue? Kirill is working on #93, he might proceed faster with that issue.

TimofeevAlex, 2017-11-16:
I'm going to finish this weekend.

closed 2017-11-21 by pavelkryukov