MIPT-ILab/mipt-mips #73: Fix factorial trace (Funcsim)

pavelkryukov, 2017-04-11:
Currently factorial fails on instruction 62:

```
./perf_sim -b ../tests/samples/factorial.out -n 100 -d -f

jr $ra
perf_sim: ../func_sim/func_memory/func_memory.cpp:79: uint64 FuncMemory::read(uint64, short unsigned int) const: Assertion `check( addr)' failed.
```

gkorepanov, 2017-04-11:
The solution seems to be straightforward: I just have to add `dst` setting into `initR` and `initJ`:
```
dst = REG_NUM_RA
```

The problem is that it shouldn't be done for `j` and `jr` instructions, only for `jal` and `jalr`. So I have to discriminate them somehow. But now `initJ, initR, initI` are discriminating instructions only by operation type without looking at funct or opcode. So I had better add new operation type like

```
OUT_J_JUMPL // jal
OUT_J_JUMP  // ja
OUT_R_JUMPL // jalr
OUT_R_JUMP  // jr,
```
hadn't I?

pavelkryukov, 2017-04-11:
Right.

closed 2017-04-11 by pavelkryukov

pavelkryukov, 2017-04-26:
Instruction JALR was not implemented correctly: it can store address not to RA only, but to any register specified in `rd` bits. Fixed now.

PR #74 (2017-04-11): Fixed jal not saving PC to ra register
This fixes #73. Now `jal` and `jalr` instructions are saving `PC + 4` into `$31` register