MIPT-ILab/mipt-mips #77: Convert portstest.cpp to GTest

pavelkryukov, 2017-04-11:


pavelkryukov, 2017-10-30:
@denislos Since you are familiar with GoogleTest environment, I advise you to continue this this task

denislos, 2017-11-03:
@pavelkryukov I think there is a problem here. miss_rate_sim.cpp and portstest.cpp are not actually unit tests. miss_rate_sim.cpp merely evaluates a miss rate in dependence on cache parameters and portstest.cpp is an example of using ports. So I consider this task as creating unit tests for cache and ports with GTest from scratch.

pavelkryukov, 2017-11-03:
Not exactly. They are tests, but not unit tests.

1. The idea is to run `miss_rate_sim` and check that its result is the same as it ever was. It is important as one may broke cache replacement and see no functional problems ("everything works") and even no performance problems (as caches are not modeled yet). Trivial tests of cache (read/write/update/replace) exist in BPU tests already, you may copy them here.

2. No, `portstest` is not just an example: if something was broken inside ports library, the program will crash. You have to wrap possible crash cases with GTest brackets + add trivial tests if you wish.

pavelkryukov, 2017-11-03:
@denislos Just FYI — you can full synchronize your fork by following sequence:

```bash
git reset --hard upstream/master
git push -f origin master
```

pavelkryukov, 2017-11-16:
@denislos Are you investigating this or may I release your assignment?

denislos, 2017-11-16:
Yes,  I'm in the process.

denislos, 2017-11-20:
I will release my assignment and work on #221 and #225 because I think it should be solved first. If no one assigns himself I will also proceed with this task at the same time.

pavelkryukov, 2017-11-20:
Labeling as 2nd semester as it is related to caches.

closed 2017-12-04 by pavelkryukov