MIPT-ILab/mipt-mips #87: Add code style githooks

pavelkryukov, 2017-04-20:
Firstly, we have to prohibit following patterns:

* trailing spaces
* try/catch
* typedef
* using namespace
* dynamic_cast

Then, we may add code style checks.

I suggest to use new repository to keep these scripts.

inedostoev, 2017-10-30:
I try ti solve this problem.
Here my solution https://github.com/inedostoev/gitHooks

pavelkryukov, 2017-10-31:
Actually I meant to use tune some existing tool, like c++lint or Vera. Do you have an experience with them? 

Questions to your tool:
1. What about trailing spaces?
2. Do I understand correctly this script runs on client side? Could we see its result on server (e.g. for each pull request)?
3. ```elif word == "namespace":``` — not good idea as we may still use namespaces per se, We should prohibit "using namespace", but actually it is checked by Clang-Tidy already.
4. Usually code analysis tool ignore a line if there is a "NOLINT" existed in comments on the line.
5. I'd rewrite code in more pythonesque style (not sure whether "if" statement is correct, don't have python right now":

```python
    blamelist = { "\t", "try", "catch", "typedef", "dynamic_cast" }
    for word in splitBuffer:
        if word in blamelist:
            print(word + " is prohibited")
            return 1
```

5. Regular expressions might help a lot here.

inedostoev, 2017-10-31:
Unfortunately, I didn't used++lint or Vera or another. 
What I should do now? I can try to rewrite this script, or I should use existing tool?

pavelkryukov, 2017-10-31:
I'd focus on integration of your script. After that it would be easy to add some external tool.

pavelkryukov, 2017-10-31:
By integration I mean that your repository (please rename it to something more meaningful, like `codeguidelines`) is added to MIPT-MIPS repository as a submodule and hooks in MIPT-MIPS are tuned to use your script. It would be nice if you describe this process in README.md so we  may include it to their repositories as well.

inedostoev, 2017-10-31:
I understand. 
To sum up, I should add check trailing spaces and using namespace. And delete analysis of comment, whet this key word is in comment, right?

pavelkryukov, 2017-10-31:
Yes.

What will your tool report if I add the following code:

```c++
mutex.lock(); // try to lock the mutex here
```

pavelkryukov, 2017-10-31:
> Do I understand correctly this script runs on client side? Could we see its result on server (e.g. for each pull request)?

As I see, you run it pre-commit. Actually such check is very annoying, as people do a lot of temporary commits, Could you put it make it pre-push at least?

pavelkryukov, 2017-10-31:
Honestly, the task has a low ROI, there are many things to do, but the value is questionable, so it costs only two points. I've added hoping there would be a person who has some experience with such tools.
I advise you to do some other task, we need to implement more MIPS instructions.

pavelkryukov, 2017-11-06:
@inedostoev One more tool is Clang-Format. As I see, it is highly customizable: https://github.com/alexseppar/RISC-V-fsim/blob/master/.clang-format

inedostoev, 2017-11-29:
I try to write code style checker, and my checker find a lot of trailing spaces.
This is a log -> [mips.log](https://github.com/MIPT-ILab/mipt-mips/files/1513009/mips.log)
Should I have to correct these errors?

pavelkryukov, 2017-11-29:
Yes, please.

pavelkryukov, 2017-11-29:
> Testing code style of file: docs/favicon.ico

It is a binary file, they must be skipped

> Testing code style of file: 4b825dc642cb6eb9a060e54bf8d69288fbee4904i.c

That file is obviously not tracked by Git, so it must be ignored.

inedostoev, 2017-11-29:
https://github.com/inedostoev/codeguidelines
I tried to deal with с++lind and vera, but i understand, that these static code analyzers don't check all needed patterns. Clang-format changed code style, but it doesn't check needed patterns too. .clang-format example is in the repository. That's why I rewrite my python script. I add `pre-push` and `pre-commit`, and user can choose best hook for him. 
I add color output, to make it clear, what is wrong. Examples -> 
[image]
[image]
And i find using namespace and try/catch.

inedostoev, 2017-11-29:
If all will be good, I can add as submodules and delete trailing spaces

pavelkryukov, 2017-11-30:
Ok, let's try to integrate it.

pavelkryukov, 2017-12-01:
Closing this with 2 points to Kirill.

Actually there is a lot of work left, hope someday a person of higher expertise would level up that to product level.

closed 2017-12-01 by pavelkryukov