MIPT-ILab/mipt-mips #91: Split BTB and BP

pavelkryukov, 2017-04-23:
_TODO: provide a description_

YanLo, 2019-01-31:
Hi! There is what I've found out:
1. Direct branches are caused by unconditional jumps, instructions "j" and "jal", ```OUT_J_JUMP```
2. They can be determined by calling ```instr.is_jump()```, defined in BaseMIPSInstr class.
But it seems I should create special method ```is_unconditional_jump```, because ```is_jumps``` checks if instr is ```OUT_J_JUMP, OUT_R_JUMP, OUT_BRANCH```, not only ```OUT_J_JUMP```.

3. So after that we may call ```is_unconditional_jump``` in decode module, right after reading instruction:     ```auto instr = read_instr( cycle);``` , line 88

4. If it is an unconditional jump, we call function ```is_misprediction``` method or somehow use methods of Branch module. It is not clear for me yet.

YanLo, 2019-01-31:
Can you please advise me a tool for Ubuntu for easier maintaining dependencies between headers. Maybe by visualisation.
For me  it sometimes quite challenging to figure out all dependencies. Thanks

pavelkryukov, 2019-01-31:
Could you please describe what do you need exactly?
If you need a tool to find where a function is defined, you need a editor with CTags installed. It may be Vim (IIRC, Vim 8 has it installed already) or Emacs, or, for something more looking like IDE, you may try Geany.

YanLo, 2019-01-31:
For example I change functions, delete some or add in one header.
After that I have to find where I should change calls of functions, change names etc.

pavelkryukov, 2019-01-31:
Most of IDEs definitely handle this. You may try free Geany or CodeBlocks, they should accept CMakeLists.txt as input.

I bet that you are bored by long compilation time, as usually people detect such error with compiler, right? You may try to use `make -j` to parallelize builds, but I agree that we should do something more radical with compilation time.

YanLo, 2019-02-01:
> For example I change functions, delete some or add in one header.
> After that I have to find where I should change calls of functions, change names etc.

It seems to me that ```.json``` file can somehow help me with this issue.
Maybe, it is possible to visualize it for easier maintaining all dependencies.

pavelkryukov, 2019-02-01:
I'm playing with [IWYU tool](https://github.com/include-what-you-use/include-what-you-use). It creates a list of includes that should be optimized, using the `.json` file. I added it to Travis flow, it outputs in the bottom of the log (https://travis-ci.com/MIPT-ILab/mipt-mips/jobs/174609514) ( `if [ $CXX == g++-7 ] && [ "$TRAVIS_OS_NAME" == "linux" ]; then` )

YanLo, 2019-02-01:
I recalled ```is_indirect_branch``` to ```is_indirect_jumps``` as they are always taken and moreover ```OUT_R_JUMP```  instructions are called  ```indirect jumps``` in Wiki.

Next I am trying to implement this code instead of  ```is_misprediction``` method
```
 50      bool is_misprediction = false;
 51 
 52     if ( instr.is_indirect_jump() )
 53         is_misprediction = instr.bp_data.target != instr.get_new_PC();
 54     if ( instr.is_direct_branch() )
 55         is_misprediction = instr.bp_data.is_taken != instr.is_jump_taken();
 56 
 57     /* handle misprediction */
 58     if ( is_misprediction)
```

but I have an error while compiling:
```home/yan/mipt-mips/simulator/modules/branch/branch.cpp: In instantiation of ‘void Branch<ISA>::clock(Cycle) [with ISA = MIPS<(MIPSVersion)0>]’:
/home/yan/mipt-mips/simulator/modules/branch/branch.cpp:82:16:   required from here
/home/yan/mipt-mips/simulator/modules/branch/branch.cpp:53:34: error: ‘const BPInterface PerfInstr<MIPSInstr<(MIPSVersion)0> >::bp_data’ is private within this context
         is_misprediction = instr.bp_data.target != instr.get_new_PC();
                            ~~~~~~^~~~~~~
In file included from /home/yan/mipt-mips/simulator/modules/branch/branch.h:12:0,
                 from /home/yan/mipt-mips/simulator/modules/branch/branch.cpp:7:
/home/yan/mipt-mips/simulator/modules/core/perf_instr.h:20:34: note: declared private here
     const BPInterface bp_data = {};
                                  ^
/home/yan/mipt-mips/simulator/modules/branch/branch.cpp:53:49: error: use of ‘auto BaseMIPSInstr<R>::get_new_PC() const [with R = unsigned int]’ before deduction of ‘auto’
         is_misprediction = instr.bp_data.target != instr.get_new_PC();
```

I think it will be better create one ```is_misprediction_decode_stage``` and ```is_misprediction_branch_stage``` with different checks to avoid such errors.

pavelkryukov, 2019-02-01:
I'd better have a immutable getter:
```c++
public:
    const auto& get_bp_data() const { return bp_data; }
```

Regarding
> `is_misprediction_decode_stage` and `is_misprediction_branch_stage`

The conceptual problem by doing it is that Instruction obtains some knowledge about pipe stages, but generally the dependency should go in opposite direction: pipe stages have all the knowledge about Instruction. If Instruction "knew" about pipe stages, then all pipe stages would knew about other pipe stages, and anything would depend on everything. That should be avoided.

pavelkryukov, 2019-02-08:
I missed the most obvious case where the flush would be optimized: warmup. If a jump or a branch was fetched for the first time, the next instruction should wait until the target is actually decoded.

YanLo, 2019-02-08:
So, at the first occurence of a jump or a branch BPU simply can't do any prediction, am I right?

YanLo, 2019-02-08:
So we must create a bubble at the decode stage in case of the first occurence of a jump or a branch in order to avoid flash, right?

pavelkryukov, 2019-02-08:
Right, 2 stages should be bubbled. At the moment, we bubble 4 stages.

YanLo, 2019-02-13:
Hi, I have studied port communication between modules and have made a conclusion that in order to implement flush at DECODE stage we need to create:
1. flush port, called ```DECODE_2_ALL_FLUSH```, which will send flush signal to BRANCH and DECODE module.
2. flush_target port, called ```DECODE_2_FETCH_TARGET```, which will send actual_target to FETCH module.
3. bp_update port,  called ```DECODE_2_FETCH```, which will call FETCH to do an update of branch prediction unit information.

YanLo, 2019-02-13:
After that, to proceed "short flushes" in decode module we will do:

```
wp_bp_update->write( instr.get_bp_upd(), cycle);
     
    /* handle misprediction */
    if ( instr.is_misprediction())
    {
        /* flushing the pipeline */
        wp_flush_all->write( true, cycle);
          
        /* notify bypassing unit about misprediction */
        bypassing_unit->handle_flush();

        /* sending valid PC to fetch stage */
        wp_flush_target->write( instr.get_actual_target(), cycle);
        sout << "misprediction on ";
    }
```

pavelkryukov, 2019-02-13:
```c++
wp_bp_update->write( instr.get_bp_upd(), cycle);
```

Does it need any condition? From the code snippet you share, branch prediction updates are sent for every instructions, not only the direct branches.

```c++
        sout << "misprediction on ";
    }
```

I assume that you have to stop decoding stage after that.

YanLo, 2019-02-13:
Ok, we need to check it like 
```
if( instr.is_jump() )
      wp_bp_update->write( instr.get_bp_upd(), cycle);
```

pavelkryukov, 2019-02-13:
> I assume that you have to stop decoding stage after that.

Perhaps I'm wrong...

YanLo, 2019-02-13:
I missed this important detail, thanks.
At this step I agree that we have to stop decode stage, because I can't see now any reasons to let decode continue processing  after flush.
But maybe it is not a trivial issue, I will study it.

pavelkryukov, 2019-02-13:
The thing is that we have the branch decoded correctly, the only problem is with the next instruction,as it was fetched from the wrong address. Therefore decoded branch may be pushed to the pipeline, but you have to verify that it carries the right target.

YanLo, 2019-02-14:
At this stage I think we should handle misprediction in decode module in this way:
1) check if instruction is ```jump```
```if ( is_jump() )``` and do bp_update.

if misprediction takes place
2) ```flush_all``` - sends flush signal to BRANCH module
3) ```bypass_unit->handle_flush```
4) ```flush_target``` - sends to FETCH stage actual target.

So, we can push decoded branch forward into pipeline, because in case of misprediction, FETCH module will receive actual target from decode.

pavelkryukov, 2019-02-14:
> 2) `flush_all` - sends flush signal to BRANCH module

Do you mean Fetch module?

> 3) `bypass_unit->handle_flush`

Could you please explain why we need that?

YanLo, 2019-02-14:
> Do you mean Fetch module?

As I see, now BRANCH module **doesn't** send flush signal to FETCH stage in case of misprediction.
But, it sends signal to EXE, MEM, DECODE and BRANCH stage.

As I understand branch module runs simultaneously with each stage, so when BRANCH stage, corresponding to MEM stage (as it implemented now) find a misprediction, it tells BRANCH stages, corresponding to FETCH, DECODE, EXE   to flush.

YanLo, 2019-02-14:
> Could you please explain why we need that?

BRANCH stage sends signal to DECODE stage to notify bypass unit when misprediction takes place.

```
    if ( instr.is_misprediction())
    {
        /* flushing the pipeline */
        wp_flush_all->write( true, cycle);
          
        /* notify bypassing unit about misprediction */
        wp_bypassing_unit_flush_notify->write( true, cycle);

        /* sending valid PC to fetch stage */
        wp_flush_target->write( instr.get_actual_target(), cycle);
        sout << "misprediction on ";
    }
```

pavelkryukov, 2019-02-14:
> BRANCH stage sends signal to DECODE stage to notify bypass unit when misprediction takes place.

Yes. My question was _why_ it's needed for Branch stage.
Is it needed if misprediction happens at decoding stage?

YanLo, 2019-02-14:
If misprediction happens at branch stage, it sends signal through port to decode stage, and decode stage do following:
```c++
    /* update bypassing unit because of misprediction */
    if ( rp_bypassing_unit_flush_notify->is_ready( cycle))
        bypassing_unit->handle_flush();
```

So, when misprediction happens at decoding stage it should do just this:

```c++
    bypassing_unit->handle_flush();
```

pavelkryukov, 2019-02-14:
You keep explaining **what** is performed (I know that perfectly) instead of trying to understand **why** it is performed.

Let me explain it for you;

If branch is mispredicted on Memory (i.e. Branch) stage:
1. There are in-progress instructions on Execution, Decode, and Fetch stages.
1. They have to be canceled since they were fetched wrongly
1. Execution and Decode stages are controlling data dependencies between instructions through logic modeled in DataBypass class.
1. Therefore, data dependency information has to be cleared out as well
1. To do that, a dedicated signal is sent to call `bypassing_unit->handle_flush()`

If branch target is mispredicted on Decode stage:
1. There is a in-progress instruction on Fetch stage
1. It has to be canceled as it is predicted wrongly
1. But, no instructions on Execution and Decode stages are affected.
1. Therefore, data dependency information is still correct and useful.

YanLo, 2019-02-18:
Hi, I have implemented some ports. As the result, one unit test fails. I don't understand what does it mean.

```
Torture_Test: Perf_Sim , MIPS 64, Core 64
-------------------------------------------------------------------------------
modules/core/t/unit_test.cpp:77
...............................................................................

modules/core/t/unit_test.cpp:86: FAILED:
  CHECK_NOTHROW( sim.run_no_limit() )
due to unexpected exception with message:
  Checker mismatch:	Checker output: 0x40017c: {1e}	or $s0, $a0, $zero	 [ $s0 =
  0x410a83 ]
  PerfSim output: 0x400168: {1e}	lui $a0, 0x41	 [ $a0 = 0x410000 ]
```

Maybe it will be better if I create a pull request, so you will see what I have done?

pavelkryukov, 2019-02-18:
I think you are able to understand what goes wrong by yourself.
Unit tests for performance simulation execute standalone traces, and you can run them with full debug output: `mipt-mips -b /path/to/traces/tt.core.core64.out -I mips64 -d`. That will help you to understand why wrong instructions are executed.

YanLo, 2019-02-20:
Can this error be caused by exe hazard, arising due my implementation in such way?
```c++
const bool is_flush = (rp_flush->is_ready( cycle) && rp_flush->read( cycle)) ||
                          (rp_flush_fetch->is_ready( cycle) && rp_flush_fetch->read( cycle));
```
It is in decode.cpp. 
```rp_flush_fetch``` is the port which is needed to flush instruction that now is on fetch stage and goes to decode, by sending flush signal from decode to itself.

Now simulator have this line:
```c++
const bool is_flush = rp_flush->is_ready( cycle) && rp_flush->read( cycle);'
```
There ```rp_flush``` receive flush signal from branch module.

pavelkryukov, 2019-02-20:
What error?

YanLo, 2019-02-20:
С уважением, Логовский Ян.

pavelkryukov, 2019-02-20:
Have you analyzed what caused the error from the debug output?

YanLo, 2019-02-21:
Yes. I have and I have a question. Why in dump of ``` mipt-mips -b /path/to/traces/tt.core64.out -I mips64 -d``` we have different from ```tt.core64.s``` instructions to execute?

For example, I see many ```bne``` instr in ```tt.core64.s``` and only one in dump.
Maybe it is caused by some build optimizations.

pavelkryukov, 2019-02-21:
No, build optimizations are disabled.
In my workspace, everything is correct: 

```bash
$> ./mipt-mips -b ../../traces/tt.core64.out -d -I mips64 | grep bne | grep wb | wc -l
146
```

YanLo, 2019-02-21:
Maybe, I have build traces in the wrong way. Maybe I should define something like mips64 in keys of building.

Because I now have
$> ./mipt-mips -b ../../traces/tt.core64.out -d -I mips64 | grep bne | grep wb | wc -l
1

1 instead of 146.

pavelkryukov, 2019-02-21:
> Maybe, I have build traces in the wrong way

Try to rebuild the trace (`cd traces && git pull origin master && make all`).

Could you please share the full log?

YanLo, 2019-02-22:
Sorry for long response. 
I have this:
```c++
~/mipt-mips/traces$ ./../simulator/mipt-mips -b tt.core64.out -d -I mips64 | grep bne
fetch   cycle 119: 0x400144: {21}	bne $zero, $zero, 539	 [ ]
decode  cycle 120: 0x400144: {21}	bne $zero, $zero, 539	 [ ]
execute cycle 121: 0x400144: {21}	bne $zero, $zero, 539	 [ ]
0x400144: {21}	bne $zero, $zero, 539	 [ ]
wb      cycle 123: 0x400144: {21}	bne $zero, $zero, 539	 [ ]
Checker mismatch:	Checker output: 0x40017c: {30}	or $s0, $a0, $zero	 [ $s0 = 0x410a83 ]
PerfSim output: 0x400168: {30}	lui $a0, 0x41	 [ $a0 = 0x410000 ]
```

pavelkryukov, 2019-02-22:
Your simulation stopped on 30th instruction, that's why only one branch was executed.
It happenede because performance simulator did a writeback of instruction different to functional simulator:
```bash
Checker mismatch:	Checker output: 0x40017c: {30}	or $s0, $a0, $zero	 [ $s0 = 0x410a83 ]
PerfSim output: 0x400168: {30}	lui $a0, 0x41	 [ $a0 = 0x410000 ]
````

Your bug is that you pushed LUI instead of OR as instruction 30.

YanLo, 2019-02-23:
Could you please explain me why I pushed LUI when it is no such instruction in tt.core64.s. How can it be?
I checked, I can't find this instruction ```lui $a0, 0x41``` even in whole repository. (in traces too)

pavelkryukov, 2019-02-23:
`lui` is a part of `li` or `la` pseudo instruction (see https://github.com/MIPT-ILab/mipt-mips/wiki/MIPS-pseudo-instructions).

pavelkryukov, 2019-02-23:
Also, you can use `mips-objdump` command of Binutils to see instructions and their placement in output file

YanLo, 2019-02-25:
I see instruction lui $a0, 0x41 in objdump of ```tt.core64.out```. But I don't see $s0, $a0, $zero instruction in this objdump. Where it can be written?

pavelkryukov, 2019-02-25:
You mean `or $s0, $a0, $zero`?
Probably, it is dumped as `move $s0, $a0` -- another pseudoinstruction.
You may run functional simulator (same options, plus `-f`) to see the "golden" sequence of instructions dumped in MIPT-MIPS format (i.e. no pseudoinstructions)

YanLo, 2019-02-25:
> You mean `or $s0, $a0, $zero`?
Yes

YanLo, 2019-02-25:
Could you please tell me where or how I can find all this instructions/manuals how to run simulator (what keys I need to set according to my exact purpose)? Thanks.

pavelkryukov, 2019-02-25:
[readme.md](https://github.com/MIPT-ILab/mipt-mips/blob/master/readme.md)

YanLo, 2019-02-26:
> Probably, it is dumped as `move $s0, $a0` -- another pseudo instruction.

Maybe, I should edit page MIPS pseudo instructions. Because analysing dump of  ```mipt-mips -b /path/to/traces/tt.core64.out -I mips64 -d -f ``` and disassembly of ```tt.core64.out```, I've found out that pseudo instruction ```move $s0, $a0``` is translated as ```or $s0, $a0, $zero```.

But in the manual page is stated that ```move $t, $s``` is expanded as ```addiu $t, $s, 0```.
It is true of course, but in my example we have another expansion.

pavelkryukov, 2019-02-26:
I don not care much — it is compiler dependent anyway, and the difference is really insignificant.

YanLo, 2019-02-26:
I've found out what happens. 29 instruction, ```jal 40017c``` (as it written in disasm file) is predicted as not taken and no flush is executed. So, PerfSim executes wrong instructions after it.

I thought about your comment
>I missed the most obvious case where the flush would be optimized: warmup. If a jump or a branch was fetched for the first time, the next instruction should wait until the target is actually decoded.

Probably, it is about my case.

YanLo, 2019-02-26:
So, I must create bubble in Fetch in the case of first occurrence of jump or branch.

pavelkryukov, 2019-02-26:
> the case of first occurrence of jump or branch.

Right. For that cases, BP predicts "not taken" and some bogus target, and we want to use that information to flush F and D stages.

YanLo, 2019-02-26:
What is the difference between first occurrence case and just misprediction?
In both cases we have wrong direction or wrong target. So, I have implemented misprediction check in decode in following way:
```c++
auto instr = read_instr( cycle);

    /* acquiring real information for BPU */
    wp_bp_update->write( instr.get_bp_upd(), cycle);

    bool is_misprediction = false;
    
    if (( instr.is_direct_jump() ) || ( instr.is_indirect_jump() ))
        is_misprediction = ( instr.get_bp_data() ).is_taken != instr.is_jump_taken();
    else if (( instr.is_direct_jump() ) || (instr.is_direct_branch() ))
        is_misprediction = ( instr.get_bp_data() ).target != instr.get_new_PC();

    /* handle misprediction */
    if ( is_misprediction)
    {
        /* flushing the pipeline */
        wp_flush_fetch->write( true, cycle);

        /* sending valid PC to fetch stage */
        wp_flush_target->write( instr.get_actual_target(), cycle);
        sout << "misprediction on ";
}
```

Probably, it doesn't work because at decode stage ```instr.is_jump_taken()``` method gives wrong info, always ```false```, as it is stated in constructor of Instr. The variable ```_is_jump_taken``` gets its actual value only in ALU module, as I've understood, finding following in ```alu.h```:

```c++
void branch_and_link( I* instr)
    {
        instr->_is_jump_taken = p( instr);
        if ( instr->_is_jump_taken)
        {
            instr->v_dst = instr->new_PC;
            instr->new_PC += sign_extend( instr) * 4;
            check_halt_trap( instr);
        }
}
---------
void branch( I* instr)
    {
        instr->_is_jump_taken = p( instr);
        if ( instr->_is_jump_taken) {
            instr->new_PC += sign_extend( instr) * 4;
            check_halt_trap( instr);
        }
}
---------
void jump( I* instr, Addr target)
    {
        instr->_is_jump_taken = true;
        instr->new_PC = target;
        check_halt_trap( instr);
}
```

P.S I've changed name ```is_indirect_branch()``` to ```is_indirect_jump()``` as the first was not quite right.

pavelkryukov, 2019-02-26:
> What is the difference between first occurrence case and just misprediction?

There is no difference. 

> at decode stage `instr.is_jump_taken()` method gives wrong info, always `false`

Yes, you are right, that's not correct. I think the good ways to fix it are:
1. rename `_is_jump_taken` to `is_branch_taken` and use it only for branches
2. rename `is_jump_taken()` to `is_taken()`; for branches, return `is_branch_taken()`, for direct branches, always return true

> P.S I've changed name `is_indirect_branch()` to `is_indirect_jump()` as the first was not quite right.

I have a huge patch affecting BaseMIPSInstr, please get sure everything is merged correctly

YanLo, 2019-02-27:
I decided to fix it in a such way:
```c++
bool is_taken() const·
      {
          if ( this->is_direct_jump() || this->is_indirect_jump() )
              return true;
          else
              return  _is_branch_taken;·
     }
```
But still PerfSim tries to execute wrong instruction.

I think I should somehow dump some bool values during simulation in oder to undestand what is still wrong.
Could you please advise me the most effective way to do it?
Maybe I should use GDB?

pavelkryukov, 2019-02-27:
I think it would be useful to print the prediction and actual result to general dump, here: https://github.com/MIPT-ILab/mipt-mips/blob/8bf5c2b750ed80571c1c04c7d90bb63ffb037424/simulator/func_sim/operation.h#L203

pavelkryukov, 2019-02-27:
Oops, sorry, it won't work -- that structure does not have the BPU information yet.
If you are OK with GDB, you may use it. But i think you can just dump "taken/not taken" on each stage.

YanLo, 2019-03-03:
Hi!
I've found out that we have the same situation with method ```get_new_PC``` as we had with method ```is_jump_taken```:

```new_PC``` is initialized in the constructor of operation as ```NO_VAL32```, so at decode stage method ```get_new_PC``` always returns ```NO_VAL32```, because it gets it's real value only in ALU section:
```c++
template<typename I, Predicate<I> p> static
    void branch( I* instr)
    {
        instr->is_taken_branch = p( instr);
        if ( instr->is_taken_branch) {
            instr->new_PC += sign_extend( instr) * 4;
            check_halt_trap( instr);
        }
}
----------------
void jump( I* instr, Addr target)
    {
        instr->is_taken_branch = true;
        instr->new_PC = target;
        check_halt_trap( instr);
}
----------------
template<typename I, Execute<I> j> static
    void jump_and_link( I* instr)
    {
        instr->v_dst = instr->new_PC; // link
        j( instr);   // jump
    }

    template<typename I, Predicate<I> p> static
    void branch_and_link( I* instr)
    {
        instr->is_taken_branch = p( instr);
        if ( instr->is_taken_branch)
        {
            instr->v_dst = instr->new_PC;
            instr->new_PC += sign_extend( instr) * 4;
            check_halt_trap( instr);
        }
}
```

So, if we want to check misprediction of target of ```direct_jump``` and ```direct_branch``` instructions at decode stage, we must change ```get_new_PC``` method.

pavelkryukov, 2019-03-03:
Let's have this:

1. Let's use `new_PC` for the actual results
2. To read predicted target, we may use this method: https://github.com/MIPT-ILab/mipt-mips/blob/73e4fd06d1cf1fe93ec8bfb047851c9d968468ec/simulator/modules/core/perf_instr.h#L29
3. To read the actual target, you need either an additional variable or some method to wrap immediates:

```c++
Addr get_decoded_target() const
{
    if ( is_branch())
        return PC + 4 + Addr{ narrow_cast<int16>(instr->v_imm)} * 4;
    if ( is_direct_jump())
        return (PC & 0xf0000000) | (v_imm << 2u);
    return 0;
}
```

pavelkryukov, 2019-03-04:
Moving a point from #613 — Yan progressed with decoupling BTB and BP.

[26 more replies]

closed 2019-03-09 by pavelkryukov