MIPT-ILab/mipt-mips #92: Implement conditional movz and movn

pavelkryukov, 2017-04-26:
MIPS IV introduces two instructions: movz and movn. You may check their behaviour on our Wiki page.
Unlike MIPS I instructions, they read 3 sources registers, so their implementation is not trivial.

pavelkryukov, 2017-11-11:
@YanLo There is a better solution than adding 3rd source to instruction. You may hack a `RF::write_dst` function like this:

```c++
        inline void write_dst( const FuncInstr& instr)
        {
            RegNum reg_num = instr.get_dst_num();
            if ( REG_NUM_ZERO != reg_num && /* check condition */)
                write( reg_num, instr.get_v_dst());
           else
                write( reg_num, read(reg_num));
        }
```

The condition should be kept in FuncInstr class. It is true for all instruction excepting `movz` and `movn` instruction with false result.

pavelkryukov, 2017-11-20:
Yan, a small **ordered** list of things included to the task besides implementation of instructions:

1. You have to uncomment new instructions in mips/t/unit_test.cpp and remove them from "testing of unsupported instructions". That change should come together with implementation, otherwise testing will fail.
1. We have a tt.core.s trace which contains all the instruction we are emulating at the moment. After pushing the changes to simulator, you have to uncomment testing of instructions you are going to implement and update the traces submodule.
1. Unfortunately, there is no other way to stop the simulation at the moment other than providing a precise amount of executed instructions. Currently it is '2013' and this number is hardcoded to func_sim/t/unit_test.cpp and perf_sim/t/unit_test.cpp. You have to increase it so tt.core.s is executed to the last instruction.

pavelkryukov, 2017-11-21:
No. we don't. We just simply will write the same value to dst (see code above).

YanLo, 2017-11-22:
Good evening! I want to test my implementation, but there is no mention about movz and movn in simulator/mips/t/unit_test.cpp, maybe I'm searching in a wrong place? And how to run this test afterwards?

pavelkryukov, 2017-11-22:
Oops, they did not exist here. Please add them.

pavelkryukov, 2017-11-23:
I added the tests, so you may proceed with commits. Firstly, open a pull request for implementation, then update `tt.core.s` and simulator unit tests.

YanLo, 2017-11-23:
Thank you, I have already done it. But a placed them straight after and instruction, is there any difference?

pavelkryukov, 2017-11-23:
It's better to use ABC order though

YanLo, 2017-11-23:
Ok, I will continue a little bit later.

YanLo, 2017-11-24:
I am counting executed instructions. Am I right that a have to count also a "syscall", and don't have to count strings like ".data", "movn_: ..." and ".text" ?

pavelkryukov, 2017-11-24:
@YanLo You may just run functional simulation and check if the latest instruction is not `nop`:
`mipt-mips -n 2013 -b ../traces/tt.core.out -f -d`

YanLo, 2017-11-24:
I had to remove movn and movz from "testing of unsupported instructions", but I can't find were should I do it, what file to modify?

pavelkryukov, 2017-11-24:
Ignore.

pavelkryukov, 2017-11-24:
You'd better fix a bug in MOVN/MOVZ implementation, test is not passing.

YanLo, 2017-11-24:
I'm sorry, but when I run tests by "make test", everything is alright. Or it is other test?

YanLo, 2017-11-24:
I asked about "unsupported" instructions because I have to build tt.core.out, but I get the message "opcode is not supported", how can I fix it?

pavelkryukov, 2017-11-24:
`make test` checks correctness of disassembly, functional correctness is verified only by running `tt.core.s`.

Have you switched build option to mips4?

pavelkryukov, 2017-11-24:
Also, always please send full error logs and command lines, telepathy is not my strength.

YanLo, 2017-11-28:
Good evening! I was ill and so incapable to solve but I will continue in immediate time.

YanLo, 2017-11-28:
I'm trying to build tt.core.out and have such errors:
```ruby
(.text+0xe00): undefined reference to `fp_d1'
```
I have found the declaration of this variable in commented block of code:
```ruby

2214 #   .data
2215 #add.d_:    .asciiz "Testing ADD.D\n"
2216 #fp_d0: .double 0.0
2217 #fp_d1: .double 1.0
2218 #fp_dm1:    .double -1.0
2219 #   .text
2220 #   li $v0, 4   # syscall 4 (print_str)
2221 #   la $a0, add.d_
2222 #   syscall
```
Should I uncomment only .data section or an entire testing block?

pavelkryukov, 2017-11-28:
It seems you touched something besides movz and movn (related to floating point operations). Please check your changes with `git diff | vim -`.

closed 2017-11-29 by pavelkryukov