MIPT-ILab/mipt-mips #93: Implement REGIMM instructions

pavelkryukov, 2017-04-26:
Some I-type MIPS instructions change behavior depending on 'rt' bits. In MIPS I they are:

* BLTZ
* BGEZ
* BLTZAL
* BGEZAL

pavelkryukov, 2017-11-17:
You may impelement the functions in meanwhile

pavelkryukov, 2017-11-20:
@inedostoev Kirill, a small list of things included to the task besides implementation of instructions:

1. You have to uncomment new instructions in `mips/t/unit_test.cpp` and remove them from "testing of unsupported instructions". That change should come together with implementation, otherwise testing will fail.
1. We have a tt.core.s trace which contains all the instruction we are emulating at the moment. After pushing the changes to simulator, you have to uncomment testing of instructions you are going to implement and update the `traces` submodule.
1. Unfortunately, there is no other way to stop the simulation at the moment other than providing a precise amount of executed instructions. Currently it is '2013' and this number is hardcoded to `func_sim/t/unit_test.cpp` and `perf_sim/t/unit_test.cpp`. You have to increase it so `tt.core.s` is executed to the last instruction.

pavelkryukov, 2017-11-28:
@inedostoev When do we expect this to be ready?

inedostoev, 2017-11-28:
I will try to do it in the coming week

pavelkryukov, 2017-12-05:
@inedostoev Do you have any code in progress? I'd like to do some refactoring of the code but I don't want to create a merge conflict for you.

inedostoev, 2017-12-05:
I don't have great progress. You can refactoring code, it will not affect on my progress

pavelkryukov, 2017-12-06:
I have pushed all the intended changes, feel free to proceed.

inedostoev, 2017-12-09:
I try to make test, and I have problems with instructions` bgezal` and `bltzal`. In normal codition it needs 2132 steps, but it fail on 480
I have init 
```cpp
       case OUT_RI_JUMP_LINK:
            v_imm = instr.asI.imm;
            src1 = static_cast<RegNum>(instr.asI.rs);
            dst = REG_NUM_RA;
            oss << " $" << regTable[src1] << ", "
                << std::dec << static_cast<uint16>(v_imm);
            break;
```
and execute 

```cpp
        template<Predicate p>
        void execute_jlr() 
        {
            v_dst = new_PC; 
            _is_jump_taken = (this->*p)();
            if ( _is_jump_taken) 
            {
                new_PC = (PC & 0xF0000000) | (v_imm << 2);
            }
        }
```

With that can be problem? I try a lot of variants, but it did'n help.

pavelkryukov, 2017-12-09:
You calculate PC with ‘jump’ formula, but the  ‘branch’ formula should be used.

pavelkryukov, 2017-12-09:
That one. Branch-and-link is an I instruction, it’s immediate has only 16 bits and encodes offset _related_ to PC + 4.

```c++
new_PC += sign_extend( v_imm) << 2;
```

inedostoev, 2017-12-09:
You mean, that it should be new_PC += sing_extend( v_imm) << 2 ?

inedostoev, 2017-12-09:
Ok, I will try

pavelkryukov, 2017-12-09:
Could you please implement these instructions as well? Sorry, but it seems I forgot to add them into original task:
* teqi
* tnei

inedostoev, 2017-12-09:
Ok, no problem

pavelkryukov, 2017-12-09:
Also, `tt.core.s` contains some so-called pseudoinstruction branches.

https://github.com/MIPT-ILab/mips-traces/blob/33a303f3b30c5d9632dcf8bc5f92a3f2aa50643b/tt.core.s#L3952

You may uncomment them as well, but it is not mandatory, I may do it later by myself.

Pseudoinstruction is an assembly-level macro which expands to one more real MIPS instructions: https://github.com/MIPT-ILab/mipt-mips/wiki/MIPS-pseudo-instructions

closed 2017-12-09 by pavelkryukov