MIPT-ILab/mipt-mips #94: Implement automated print of MIPS ISA

pavelkryukov, 2017-04-26:
Currently we have much information about ISA in FuncInstr. It might be useful to print it nicely by some command.

TimofeevAlex, 2018-03-06:
Should I implement class with only ISA and dump() that will print ISA? Or should I add another something?

TimofeevAlex, 2018-03-06:
Ok, I realize

TimofeevAlex, 2018-03-06:
I did not fully understand what the function generate has to do.  Why should it return an object MIPSInstr?

pavelkryukov, 2018-03-06:
Yes, we want to move ctor inside a new class.
It is a kind of factory of MIPSInstr's: you provide 32 bits (and PC value), it gives you an instruction. Some internal variables may control the output, i.e. emit only MIPS I instructions.

Let's change the signature a little bit:

```c++
class MIPS {
public:
    MIPSInstr generate( uint32 raw, Addr PC = 0) const;
    void dump() const;
};
```

To solve problem with encapsulation, use `friend` keyword in MIPSInstr. It prevents you from moving everything to `public` visibility:

```c++
class MIPSInstr {
// ...
    friend class MIPS;
};
```

pavelkryukov, 2018-05-04:
@TimofeevAlex If you have something in progress, please share before deadline

ivababkin, 2018-10-03:
I'd like to take this problem

pavelkryukov, 2018-10-05:
Sure. If you have questions, put them here, so @olegladin can assist you.

PR #700 (2018-11-19): Move all MIPS decoding to a separate translation unit
Fixes #94
commit d5952500 (2018-11-19): Move all MIPS decoding to a separate translation unit

Fixes #94
closed 2018-11-19 by pavelkryukov