MIPT-ILab/mipt-mips #122: Implement MARS syscalls

pavelkryukov, 2017-05-08:
There are wiki page about [MARS syscalls](https://github.com/MIPT-ILab/mipt-mips/wiki/MARS-syscalls). We are going to implement following codes:

- 1 print integer
- 4 print string
- 5 read integer
- 8 read string
- 10 exit — stops simulation.
- 11 print character 
- 12 read character
- 13 open file — this and below fully reuses host machine descriptors
- 14 read file
- 15 write file
- 16 close file
- 17 exit with code — stops simulation and returns code from main.cpp

Implementation of syscall must be performed in special C++ class MIPSCalls.
For performance simulation, syscalls are performed on WB stage to get sure they are not bogus due to mispredicted branch.

pavelkryukov, 2017-11-16:
There should be two options for FuncSim:
* to enable MARS system calls, so the interaction happens
* to ignore all system calls

The best way to implement is a polymorphic inheritance:

```c++
class MIPSCalls {
    MIPSCalls(MIPSMemory* mem); // Memory is required to read data
public:
    virtual void process_syscall(uint32 callcode,  /* other arguments */) { /* do nothing */}
};

class MARSCalls : public MIPSCalls {
// ...
};
```

pavelkryukov, 2017-11-26:
@qsuo Any news? It would be a nice feature to have it soon, as we still use hacks with instruction counting.

qsuo, 2017-11-27:
@pavelkryukov I have had some problems with my free time, so I'm going to start at this weekend

pavelkryukov, 2018-02-21:
@qsuo Any news? That feature will significantly improve our simulation:
- we will be able to run interactive traces from other education projects
- we may proceed by extending system calls by our own internal OS (mostly to have `sbrk` (`malloc`) system call

trexxet, 2018-10-10:
Should the "kernel" code be called from `execute_syscall()` in MIPSInstr class or from FuncSim on trap check?

pavelkryukov, 2018-10-10:
> "kernel" code be called from execute_syscall() in MIPSInstr class

How would you put the "kernel" to MIPSInstr class?

trexxet, 2018-10-10:
I mean just calling it. However, as I thought a bit more, executing syscall handler from MIPSInstr would be troublesome. 
I think we can replace polymorphic inheritance with a simple flag in FuncSim:
```c++
FuncSim::run() {
    ...
    if (instr.trap_type() == Trap::SYSCALL && !ignore_syscalls)
         callSyscallHandler();
    ...
}

pavelkryukov, 2018-10-10:
> I think we can replace polymorphic inheritance with a simple flag in FuncSim:

The basic motivation for polymorphic interfaces is removal of flags.
Caller does not have to know whether handler will do anything.

pavelkryukov, 2018-11-07:
IMO the best way to enable them is a `--mars` switch

pavelkryukov, 2018-11-13:
I'm adding points to your scoreboard entry as 90% of task is completed. Adding remaining calls is quite trivial now, you may proceed with them.

closed 2018-12-16 by pavelkryukov