MIPT-ILab/mipt-mips #124: Convert PC and new_PC to ports

pavelkryukov, 2017-05-10:


pavelkryukov, 2018-02-21:
@alex19999 Have you made any progress on this last year?

alex19999, 2018-02-21:
Oh, sorry, I forgot about this task. If it is necessary, I can try to do it firstly21.02.2018, 21:22, "Pavel I.

pavelkryukov, 2018-02-21:
That's a very easy task. If you finish it by Feb 26, you got 1 point, otherwise I'll do it myself.

alex19999, 2018-02-21:
Ok, I understood. I can't promise you a result, because I will work at the "Phystech" olympiad this weekend, so now I must do some homework. No matter I will try.21.02.2018, 21:36, "Pavel I.

alex19999, 2018-02-22:
Pavel Igorevich, may I do Core->IF port without any latency? Or what can I do? I have problems at this stage21.02.2018, 21:36, "Pavel I.

pavelkryukov, 2018-02-22:
Zero-latency ports are possible, but it is better to avoid them.
Why do you need them?

pavelkryukov, 2018-02-22:
> I have a problem, that Core->IF port can't read PC from cycle = 0, but it is necessary to initialize starting PC value

I think the best option is to issue a nop to the pipeline if there is no valid PC

pavelkryukov, 2018-02-22:
+ability to issue a nop to the pipeline is a direct way to implement instruction cache

alex19999, 2018-02-25:
Pavel Igorevich, hello. Can you help me?
Firstly, I add function `get_PC( Cycle cycle)`, which helps me to get valid(or not) PC

`    template <typename ISA> Addr PerfSim<ISA>::get_PC( Cycle cycle)
    {  `
     
      const bool is_flush = rp_fetch_flush->is_ready( cycle) && rp_fetch_flush->read( cycle);
      const bool is_stall = rp_decode_2_fetch_stall->is_ready( cycle) && rp_decode_2_fetch_stall->read( cycle);
      if(rp_core_2_fetch_target->is_ready( cycle)) 
          return rp_core_2_fetch_target->read(cycle);
      if ( is_flush)
          return rp_memory_2_fetch_target->read( cycle);
      else if ( !is_stall)
          if(rp_target->is_ready( cycle))
              return rp_target->read( cycle);   
          return NO_VAL32;
       } `

Then I introduce the `get_instr(Addr PC)` function, which generate  instruction, which will be sent to decode stage

`template <typename ISA>
 typename PerfSim<ISA>::Instr PerfSim<ISA>::get_instr( Addr PC)
  {`
      
     if(PC != NO_VAL32)
          return Instr( memory->fetch_instr( PC), bp->get_bp_info( PC));
      return  Instr( FuncInstr(0x0u),  bp->get_bp_info( PC));
     }
`
 

In fetch stage I have

 
  ` auto PC = get_PC( cycle);` 
  `auto instr = get_instr( PC);`
 ` wp_fetch_2_decode->write( instr, cycle);`

But in the result there is only mismatch, for example

Mismatch: 
Checker output: 0x4000d0: or $t1, $zero, $zero	 [ $t1 = 0x0 ]
PerfSim output: nop 

Could you help me?

pavelkryukov, 2018-02-25:
1. For invalid PC, use 0 instead of NO_VAL32 as there is some probability that real program will use 0xdeaddead address.
2. Then you have an instruction with 0 bytes and 0 PC. It should be a special case;
```c++
MIPSInstr bool is_bubble() const {
    return is_nop() && PC == 0;
}
```
3. Bubbles should be ignored by WB stages:
```c++
    auto instr = rp_memory_2_writeback->read( cycle);

    /* perform writeback */
    rf->write_dst( instr);

    /* NEW CODE HERE */
    if ( instr->is_bubble())
        return;

    /* check for traps */
    instr.check_trap();

    /* log */
    sout << instr << std::endl;

    /* perform checks */
    check( instr);

    /* update simulator cycles info */
    ++executed_instrs;
    last_writeback_cycle = cycle;
```

alex19999, 2018-02-25:
I added your code and it started working perfectly, but only in one instruction, then I have infinite cycle in writeback. I will try to fix it

closed 2018-02-27 by pavelkryukov