MIPT-ILab/mipt-mips #130: Implement traps

pavelkryukov, 2017-05-16:
Many of MIPS instruction may perform traps — internal exceptions:
* Incorrect load/store address
* Incorrect return address
* Unknown instruction
* Explicit trap (`teq, tne` and similar instructions)
* etc.

There is a good explanation of traps handling: http://www.cs.iit.edu/~virgil/cs470/Labs/Lab7.pdf

DavidRabbitson, 2018-10-27:
Currently I'm trying to fix CMake problems and solve some IDE issues. Hope it won't last long

DavidRabbitson, 2018-10-30:
Finally, I compiled simulator successfully. And now I'm working on my C++ skills.

DavidRabbitson, 2018-11-05:
Don't have time to work on this assignment this week because of huge amount of homework.

pavelkryukov, 2018-11-23:
@DavidRabbitson Since Vsevolod is completing test suite for MIPS trap handling, I propose you to continue his job on #530, while he shall do this (#130) task instead. Feel free to express objections if you have.

DavidRabbitson, 2018-11-24:
@pavelkryukov Ok, no problem, I'll try.

pooh64, 2019-04-18:
Addition overflow now looks like:
https://github.com/MIPT-ILab/mipt-mips/blob/3e500b950d4dad66a789a9cfe593646f1fd23726/simulator/func_sim/alu.h#L160-L166
addi instruction must not modify destination register.

So can I:
Implement this behavior inside a func_sim (directly in addition_overflow function or new "restore_dst" function to link it with addi in mips_instr)?
Add set_v_dst to test this case?

pavelkryukov, 2019-04-18:
Yes, sure. All behavior should be compliant with MIPS ISA specification. Better to have the unit tests for that.

pooh64, 2019-04-19:
This unit test:
https://github.com/MIPT-ILab/mipt-mips/blob/91325abab7b0b00d5caae42eefa8140090a0c71b/simulator/modules/core/t/unit_test.cpp#L76-L86
catches exception from here, called from mem.clock():
https://github.com/MIPT-ILab/mipt-mips/blob/91325abab7b0b00d5caae42eefa8140090a0c71b/simulator/memory/memory.h#L151-L153
after integer_overflow trap in alu happens while executing this step:
https://github.com/MIPT-ILab/mipt-mips/blob/91325abab7b0b00d5caae42eefa8140090a0c71b/simulator/func_sim/func_sim.cpp#L35-L48
called from PerfSim.clock_tree() -> writeback.clock()

I'm a little confused, step is called from writeback.clock(), but exception with zero address happens in mem.clock(), as if previous instruction trap was not handled. Where should I handle "unhandled trap"?

pavelkryukov, 2019-04-19:
First of all, stores to 0 address should never happen in our traces. Usually it means that the trace is executed wrongly, so exception string should be not "unhandled trap" but "critical error". Please run the trace by simulator and check what causes the unexpected behavior.

pooh64, 2019-04-22:
Mars exceptions list:
ADDRESS_EXCEPTION_LOAD (4)
ADDRESS_EXCEPTION_STORE (5)
SYSCALL_EXCEPTION (8)
BREAKPOINT_EXCEPTION (9)
RESERVED_INSTRUCTION_EXCEPTION (10)
ARITHMETIC_OVERFLOW_EXCEPTION (12)
TRAP_EXCEPTION(13)
DIVIDE_BY_ZERO_EXCEPTION (15)
FLOATING_POINT_OVERFLOW (16)
FLOATING_POINT_UNDERFLOW (17)

I think our TrapType must have these all.

https://github.com/MIPT-ILab/mipt-mips/blob/65b9f73ceaa3800ed808f5eb7c73108883408eef/simulator/func_sim/trap_types.def#L1-9

UNALIGNED_ADDRESS must be splitted in ADRESS_EXCEPTION_LOAD/STORE
UNSUPPORTED_SYSCALL probably must be handled by our kernel module.
+SYSCALL (call kernel from this point)
+RESERVED_INSTR
+DIVIDE_BY_ZERO (but it's not generated by original div and other instructions, what to do?)
+FP traps, but there are no FPU instructions implementation now.

Maybe we must have some enum with reaction types in FuncSim? It's much better than Value<std::string>.

pavelkryukov, 2019-04-22:
>  It's much better than Valuestd::string.

Why? Strings are more readable and introduce less coupling between classes. Once you provide a string interface, you don't have to update it any longer (e.g. by adding new enum variations).

> I think our TrapType must have these all.

Sure.

> (but it's not generated by original div and other instructions, what to do?)
What do you mean by original?

pooh64, 2019-04-22:
> What do you mean by "original"?
According to MIPS ISA.

pooh64, 2019-04-22:
> UNSUPPORTED_SYSCALL probably must be handled by our kernel module.
> +SYSCALL (call kernel from this point)

And what to do with this? Call the kernel right inside FuncSim::run if trap == SYSCALL?
Also Mars SYSCALL_EXCEPTION, I think it can't occur in kernel module now.

pavelkryukov, 2019-04-23:
> Call the kernel right inside FuncSim::run if trap == SYSCALL?

I think that would be the correct way.

> Also Mars SYSCALL_EXCEPTION, I think it can't occur in kernel module now.

At the moment, we throw a C++ exception if something goes wrong and reexecute exception once again. Let's stick to that behavior for simplicity.

https://github.com/MIPT-ILab/mipt-mips/blob/a8d53f2445dba1ca41ad416f0bf6717b7749c1de/simulator/func_sim/func_sim.cpp#L69-L79

commit b8683e5f (2019-05-06): Handle traps in functional simulation (#984)

Fixes #130
commit 684c62be (2019-05-06): Handle traps in functional simulation (#984)

Fixes #130
commit a9fe0251 (2019-05-06): Handle traps in functional simulation (#984)

Fixes #130
commit 1e386aaf (2019-05-06): Handle traps in functional simulation (#984)

Fixes #130
closed 2019-05-06 by pavelkryukov