MIPT-ILab/mipt-mips #132: Implement unaligned memory accesses: LWL, LWR, SWL, SWR

pavelkryukov, 2017-05-19:


pavelkryukov, 2018-04-16:
Hi @AndrewSultan 

I submitted a code review with updated infrastructure, which should support LWL instruction as well. I expect you to uncomment LWL instruction code in `tt.core.s` and check that my solution works. If not, please propose fixes.

Additionally, `tt.core.s` contains tests for ULW pseudo-instruction which is just LWL and LWR instructions in the same box.

Then, you may proceed by implementing SWL, SRW, and USW using the similar infrastructure.

I would highly appreciate if you create a Wiki page about those instructions, explaining them in a way you find most comfortable, however that is not mandatory.

pavelkryukov, 2018-04-16:
@AndrewSultan 

Additionally, it would be nice if you implemented unit tests for masked writes to RF.
See `simulator/func_sim/rf/t/unit_tests.cpp` for example and [our Wiki manual](https://github.com/MIPT-ILab/mipt-mips/wiki/Introduction-to-Test-Driven-Development)

AndrewSultan, 2018-04-17:
First of all, we should add an addition condition in if in mipt_instr.cpp: 
`if ( operation == OUT_I_LOAD || operation == OUT_I_LOADR || operation == OUT_I_LOADL)`
Secondly, change is_loadlr function:
`return (operation == OUT_I_LOADR || operation == OUT_I_LOADL) ? 1 : 0;`
But the tests still do not pass.

AndrewSultan, 2018-04-17:
0x402484: addiu $v0, $v0, 0xb9b4	 [ $v0 = 0x42b9b4 ]
0x402488: or $v1, $zero, $zero	 [ $v1 = 0x0 ]
num            = 0
val before lwl = 3020100
mask           = ffffffff
val = ( val & mask) | ( static_cast<RegDstUInt>(get_value( num)) & ~mask);
val after  lwl = 3020100
0x40248c: lwl $v1, 0x0($v0)	 [ $v1 = 0x3020100]
0x402490: bne $v1, $zero, 289

pavelkryukov, 2018-04-17:
> if ( operation == OUT_I_LOAD || operation == OUT_I_LOADR || operation == OUT_I_LOADL

I performed it in a git branch. Please review the code in #397

pavelkryukov, 2018-04-18:
LWL is fixed — please proceed with stores, documentation, and unit tests for masked writes.

AndrewSultan, 2018-05-02:
I'm going to write wiki a little bit later.

pavelkryukov, 2018-05-02:
What about stores? I have not seen them operating

commit 0d29370c (2018-05-04): Add more stressful SWL tests (fixes #132)
closed 2018-05-04 by pavelkryukov