MIPT-ILab/mipt-mips #137: Add MIPS32 clo and clz instructions

pavelkryukov, 2017-05-23:
`clo` and `clz` are MIPS32 instructions used to calculate # of 1 and 0 bits in register respectively. They do not belong to MIPS IV ISA but usually modeled as they do not contradict MIPS IV.

To enable them, you have to introduce new format type FORMAT_MIPS32 as MIPS32 uses `011100` opcode for extension.

For C++ implementation you may check this classical manual page: https://graphics.stanford.edu/~seander/bithacks.html

pavelkryukov, 2017-11-21:
a small list of things included to the task besides implementation of instructions:

1. You have to uncomment new instructions in mips/t/unit_test.cpp and remove them from "testing of unsupported instructions". That change should come together with implementation, otherwise testing will fail.
1. We have a tt.core.s trace which contains all the instruction we are emulating at the moment. After pushing the changes to simulator, you have to uncomment testing of instructions you are going to implement and update the traces submodule.
1. Unfortunately, there is no other way to stop the simulation at the moment other than providing a precise amount of executed instructions. Currently it is '2013' and this number is hardcoded to func_sim/t/unit_test.cpp and perf_sim/t/unit_test.cpp. You have to increase it so tt.core.s is executed to the last instruction.

pavelkryukov, 2017-11-29:
@TimofeevAlex Do we have a progress here?

TimofeevAlex, 2017-11-29:
Not yet, I want to start on Saturday

TimofeevAlex, 2017-12-01:
I think that it not very useful if we will create new unordered map and new function for initialize two new instructions from MIPS32 as MIPS32 have a wide variety of instructions and for new instructions we will have to change a lot so I suggest adding clo and clz in R-instructions with a new type of operation especially they have the same structure as R-instructions, but I faced on one problem: we used funct as key, but 0x20,0x21 already busy, what should I do?

pavelkryukov, 2017-12-01:
No, why? I don’t see a problem with creating a new map. You may rename it to instrs011100 or instrsSpecial or whatever else. I would be much better than create workarounds to keep everything in one place (you made a big effort to do the opposite thing)

pavelkryukov, 2017-12-01:
MIPS32 defines 011100 as SPECIAL2, that may be the best name for it.

TimofeevAlex, 2017-12-01:
Okey, should I implementation new function for initialize and in function init Format() add case 0x1C (011100) or 0x2?

TimofeevAlex, 2017-12-01:
you said that somewhere in the meik you need to change to MIPS32 but I can't find where

TimofeevAlex, 2017-12-01:
I'm almost done but my code does not pass one unit test and I don't find error

pavelkryukov, 2017-12-02:
What is the output?

TimofeevAlex, 2017-12-02:
Output(clz analogically):

```
**mips/t/unit_test.cpp:71: Failure
Expected equality of these values:
  FuncInstr(0x71208821).Dump()
    Which is: { '0' (48, 0x30), 'x' (120, 0x78), '7' (55, 0x37), '1' (49, 0x31), '2' (50, 0x32), '0' (48, 0x30), '8' (56, 0x38), '8' (56, 0x38), '2' (50, 0x32), '1' (49, 0x31), '\t' (9), 'U' (85, 0x55), 'n' (110, 0x6E), 'k' (107, 0x6B), 'n' (110, 0x6E), 'o' (111, 0x6F), 'w' (119, 0x77), 'n' (110, 0x6E) }
  "clo $s1, $t1"
    Which is: 0x4722eb
[  FAILED  ] Func_instr_disasm.Process_Disasm (0 ms)
[----------] 2 tests from Func_instr_disasm (0 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 2 test cases ran. (2 ms total)
[  PASSED  ] 2 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] Func_instr_disasm.Process_Disasm

 1 FAILED TEST
Makefile:93: ошибка выполнения рецепта для цели «mips/t/test.run»
make: *** [mips/t/test.run] Ошибка 1**
```

pavelkryukov, 2017-12-02:
Yes, there is an error in your code, disassembly is not generated for new instructions. Have you handled them?

TimofeevAlex, 2017-12-02:
Apparently not. How can this be remedied?

pavelkryukov, 2017-12-02:
Ctor calls one of these methods `FuncInstr::initR`, `FuncInstr::initI`, FuncInstr::initJ`. Depending on `FuncInstr::operation` they set source registers, destination registers and generate disassembly. You have to handle `clo` and `clz` in one of these methods or create a new method.

TimofeevAlex, 2017-12-02:
I have created new method therefore I think there is a mistake there

closed 2017-12-03 by pavelkryukov