MIPT-ILab/mipt-mips #147: Add Git Submodules manual to Wiki

pavelkryukov, 2017-05-27:


inedostoev, 2017-09-29:
Add wiki page as Git Submodules

pavelkryukov, 2017-09-29:
A nice page!
I see you are used some text from more detailed GitHub manual — can you please add a link to it or some other materials you used?
I would like to see a description of the following flow: if I want to change some trace and then to fix some code in the main repository, what should I do?

pavelkryukov, 2017-09-29:
Please, add links to the page in _Sidebar and Home

pavelkryukov, 2017-09-30:
> GNU Binutils/GoogleTest version update

Just to clarify, we usually update not to the latest tagged version (like binutils_2.29_1).

inedostoev, 2017-09-30:
Corrected your comments, add links to sidebar and page home, near link about using git.
Try to change binutils, but I can't build it in Travic Ci
This is log: https://travis-ci.org/inedostoev/mipt-mips

pavelkryukov, 2017-09-30:
They've added some code which Clang does not like. We may switch Binutils build back to GCC7 by following steps:
* Reverting my commit 29ca18d56493daea3476fa02d4fd2fbbb425ee5e
* Adding gcc@7 to OS X requirements:

```diff
- if [ "$TRAVIS_OS_NAME" == "osx" ]; then brew install libelf zlib; fi
+ if [ "$TRAVIS_OS_NAME" == "osx" ]; then brew install libelf zlib gcc@7; fi
```

Actually it is a good opportunity for your to report the bug to GNU Binutils community.

pavelkryukov, 2017-09-30:
Ok, it seems there is an alternative solution to remove warning for OS X: add `--disable-werror` to configure script

pavelkryukov, 2017-09-30:
BTW, GNU folks claim that they won't fix the problem with `sbrk` deprecation: https://sourceware.org/bugzilla/show_bug.cgi?id=16644

But the problem below should be reported.
`bfd.c:519:21: error: passing an object that undergoes default argument promotion to 'va_start' has undefined behavior [-Werror,-Wvarargs]`

inedostoev, 2017-09-30:
Update binutils to realese ver. 2.29_1
GoogleTest was already updated to 1.8, it is last version

pavelkryukov, 2017-09-30:
Ok, thanks. What about
> I would like to see a description of the following flows:
> ...
> GNU Binutils/GoogleTest version update
?

inedostoev, 2017-09-30:
Oh, sorry. I thought, that it was task only to update this submodule.
Now I add description of update Binutuls, GoogleTest and the rest of the submodules.

pavelkryukov, 2017-09-30:
Great, thank you!

closed 2017-09-30 by pavelkryukov