MIPT-ILab/mipt-mips #165: Use CMake instead of Make

pavelkryukov, 2017-09-23:
It would be nice to replace GNU Make makefile with CMake makefile. In addition, you have to write a Wiki page explaining CMake as good as [this page](https://github.com/MIPT-ILab/mipt-mips/wiki/Makefiles) explains GNU Make

pavelkryukov, 2017-11-07:
@col14m Hope you are well. Do you have any news regarding CMake?

pavelkryukov, 2017-11-16:
@col14m You are working for a month here, it would be nice if you shared some code at some branch. Actually I think we may keep both CMake and Makefile for some time.

pavelkryukov, 2017-11-22:
@col14m Any news, any updates?

pavelkryukov, 2017-12-07:
@SoshinK The only one week is left before deadline. Do you estimate to have everything working beforehand?

SoshinK, 2018-03-01:
Good day(night?), Pavel. You probably thought that I gave up. In fact, I am still in. By that moment, I made Cmake for simulator(and a short buildscript for fast launch). But I have some problems with unit test for infra/config and with clang-tidy(It says 'stddef.h' not found). What should I do? Here solution(or should I make a pull request?): https://github.com/SoshinK/mipt-mips

pavelkryukov, 2018-03-01:
That's great!

Please merge with the latest version and integrate CMake to Travis. Then do a pull request so I can see output.

pavelkryukov, 2018-03-02:
Please open a pull request so I can comment your code

SoshinK, 2018-03-02:
I still didnt cope with travis.yml

pavelkryukov, 2018-03-07:
The page: https://github.com/MIPT-ILab/mipt-mips/wiki/cmake

SoshinK, 2018-03-12:
I havent finished wiki about CMake yet

SoshinK, 2018-03-12:
There the first part https://github.com/MIPT-ILab/mipt-mips/wiki/CMake

pavelkryukov, 2018-03-12:
OK, looks good. I'll make some fixes (spelling, grammar) if needed.

Could you please add a section "Further Reading" to list additional sources of information (e.g. the one you used)?

pavelkryukov, 2018-03-15:
@SoshinK What about README changes?

closed 2018-03-15 by pavelkryukov