MIPT-ILab/mipt-mips #201: Generalize PerfSim and FuncSim with ISA

pavelkryukov, 2017-10-18:
The idea is to implement:

```c++
struct MIPS
{
    using FuncInstr = MIPSInstr;
    using RF = MIPSRF;
    using Memory = MIPSMemory;
    using Syscalls = MIPSSyscalls;
    uisng Traps = MIPSTraps;
};
```

and afterwards:
```c++
template<typename ISA>
class PerfSim : public ISA
{
// ...
};

using PerfMIPS = PerfSim<MIPS>;
```

It will significantly simplify implementation of RISC-V and other architectures.

pavelkryukov, 2018-01-30:
No, I think we should not affect folder structure. Everything ISA-related is located in <isa> folders already (at the moment we have only MIPS).

Are you familiar with idea of C++ templates? Briefly, they can reduce a lot of copy-paste code. For example, you have some polynomial function, but it should be applied to any type of number. The naive approach is to write a lot of functions:

```c++
int func(int x) { return 2 * x * x + 3 * x + 1; }
double func(double x) { return 2 * x * x + 3 * x + 1; }
Complex func(const Complex& x) { return 2 * x * x + 3 * x + 1; }
```

With C++ templates you may leave the type undefined and let compiler to do the substitution (type deduction):

```c++
template<typename T> // T is a name of type
T func(const T& x) { return 2 * x * x + 3 * x + 1;}
```

That is called 'meta-programming'. In C++, classes can also be templated, like std;:vector. 
In our case T will determine ISA, so instead of having two PerfSim's (MIPSPerfSim and RISCVPerfSim), we will have one PerfSim:

```c++
template<typename ISA>
class PerfSim : ISA {
     RF* rf; // RF is ISA::RF. If ISA is MIPS, then RF is MIPSRF.
};
```

Then we may use the simulator we want:
```c++
if (isa == "mips") {
    simulator = new PerfSim<MIPS>(...);
}
else if (isa == "riscv") {
    simulator = new PerfSim<RISCV>(...);
}
```

alex19999, 2018-01-30:
I know about meta-programming, but thank you for details) So, I don't need to change structure of folders, so may I paste these strings in perf_sim.h? And now we have MIPS class, but we want to determine struct MIPS. Our compiler will be against. Sorry for many questions, brain is after session, I think, that you understand)

pavelkryukov, 2018-01-30:
> Sorry for many questions, brain is after session, I think, that you understand

I'm also a little bit out of context, so your questions are helpful to me.

> we have MIPS class, but we want to determine struct MIPS

Yes, good observation. You may rename existing class MIPS to FuncSim and do a stand-alone pull request for that (making a lot of small commits is a good practice IMO).

> I don't need to change structure of folders, so may I paste these strings in perf_sim.h? 

I'd create a standalone file in `mips` directory, `mips.h` to have all the struct inside.

alex19999, 2018-01-30:
Thank you very much. I will try.

- [ I'm also a little bit out of context, so your questions are helpful to me.]
I hope that this task is some kind of warm-up, then I want to try more difficult tasks. Thus, you will have a lot of questions)

alex19999, 2018-01-31:
Pavel Igorevich, May I delete perf_sim.h at the end of my work?(PerfMips is defined in this file, but we will do it in mips.h, so this file is unuseful). And, forgive me, if I am not right, but here:

- struct MIPS
{
    using FuncInstr = MIPSInstr;
    using RF = MIPSRF;
    using Memory = MIPSMemory;
    using Syscalls = MIPSSyscalls;
    using Traps = MIPSTraps;

- };

"Using"  is not properly used(MIPSInstr,MIPSRF and so on are aliases, but RF, FuncInstr and MIPSMemory(I want rename it in just "Memory") are real classes. So, can I rename MIPSMemory? And can you tell me something about Syscalls and Traps? We don't have these types in our simulator, compiler  "swears". Thanks, sorry for troubling you

pavelkryukov, 2018-01-31:
No we won't define PerfMIPS in 'mips.h'. 'mips.h' should contain all the aliases to MIPS ISA (instruction, register file, memory, I/O) classes. These classes should contain 'MIPS' in their names, so, when PerfSim<MIPS> is instanced, it will get correct classes (”RF is MIPSRF, FuncInstr is MIPSInstr etc.“).

Could you please send your current chances to a review?

alex19999, 2018-01-31:
Thank you for comments. Sorry, not now, may be later or next day. I changed many details in my computer and reinstall some programms, so now I have some problems with git and can't send current chances. I decided, that I will do my work and then will fix problems with git.

alex19999, 2018-02-01:
Pavel Igorevich, hello. There is one question about "using": If I write this code 

- [struct MIPS
{
    using FuncInstr = MIPSInstr;
    using RF = MIPSRF;
    using Memory = MIPSMemory;
    using Syscalls = MIPSSyscalls;
    using Traps = MIPSTraps;
};

template<typename ISA>
class PerfSim : public ISA
{
// ...
};

using PerfMIPS = PerfSim<MIPS>; ] 

in mips.h file, we will have problem, connected with redefenition PerfMips, isn't it?

pavelkryukov, 2018-02-01:
Why do you need to define PerfSim/PerfMIPS in "mips.h"?

pavelkryukov, 2018-02-01:
(also, your code is not a C++ code)

alex19999, 2018-02-01:
We need to generalize PerfSim and FuncSim, so I think, that we need separate file for this. So, where can I  put class PerfSim?(In perf_sim.h, where now PerfMips is situated?)

pavelkryukov, 2018-02-01:
Yes, perf_sim.h and perf_sim.cpp should contain **generalized** simulator which may work with any ISA.
The same thing is for func_sim.

alex19999, 2018-02-01:
Ok, I misundestood a bit, Thank you)

pavelkryukov, 2018-02-01:
I expect you'll get a linkage error when you templatize PerfSim class and methods. To fix it, add the following line into perf_sim.cpp:
```c++
template class PerfSim<MIPS>;
```

It would tell the compiler to instantiate PerfSim template for MIPS parameter.

The alternative solution is to move everything into header file, but that would decrease compilation speed.

alex19999, 2018-02-01:
Ok, I will add, thanks

pavelkryukov, 2018-02-01:
Also, I suspect that inheriting ISA won't resolve name resolution problem: 
```c++
struct Dummy {
    using B = int;
};

template<typename T>
class A : public T // may we inherit T?
{
public:
    B func(const B& x) const { return x; } // what is B?
};

int main() {
    A<Dummy> obj;
    obj.func(2);
}
```

We should do this instead:
```c++
struct Dummy {
    using B = int;
};

template<typename T>
class A
{
    using B = typename T::B;
public:
    B func(const B& x) const { return x; }
};

int main() {
    A<Dummy> obj;
    obj.func(2);
}
```

closed 2018-02-08 by pavelkryukov