MIPT-ILab/mipt-mips #214: Implement MIPS III doubleword arithmetic instructions

pavelkryukov, 2017-11-08:
MIPS III introduced a lot of doubleword instructions which operate with 64 bit extended registers. They are rarely used as most of MIPS HW is 32-bit, but we might be interested to keep them as well

* `dadd`
* `daddi`
* `daddiu`
* `daddu`
* `ddiv`
* `ddivu`
* `dmult`
* `dmultu`
* `dsll`
* `dsllv`
* `dsll32`
* `dsra`
* `dsra32`
* `dsrav`
* `dsrl`
* `dsrl32`
* `dsrlv`
* `dsub`
* `dsubu`

pavelkryukov, 2017-12-04:
@TimofeevAlex. That is not as easy as `clo`/`clz`!

1. The best start is just to add the instructions to disassembly. I expect to have a pull request here. It would be very nice if you add the tests to `mips/t/unit_test.cpp` as well.
2. After that, we have to update our RF (and internal FuncInstr variables) as it has only 32-bit registers at the moment. If we switch the type, to 64-bit, some 32-bit arithmetic could become broken — so we have to check it as well. I expect next pull request here.
3. Then, you may implement the instructions. The flow is the same as in #137: uncomment them in `tt.core.s` and check if everything works correctly.

TimofeevAlex, 2018-02-16:
Well, should I change the unit tests  and create the instructions after?

pavelkryukov, 2018-02-17:
Disassembly and unit tests should be checked in simultaneously, otherwise you get either non-working tests or no tested code.

pavelkryukov, 2018-02-17:
Then, I think you need to add "is64" Boolean variable to ISAEntry and MIPSInstr class. If variable is set to false, then values read from RF and written to RF is casted to uint32; otherwise 64-bit types are used. That would allow you to re-use code of 32 instructions (e.g. use `execute_addiu` to execute DADDIU instruction).

inedostoev, 2018-03-26:
I and Alex have a question. We together don't understand, how to realize 128-bit division? As we know, at first we write value at v_dst, after that first and second part put in hi/lo register. We change shift in rf.h and in division, but it doesn't help. What you should pay attention to?

pavelkryukov, 2018-03-26:
Probably it is a side effect of data bypass by @denislos, as his code does one more `>> 32` shift.
I suggest to disable them for div/mult instructions (either for all or for yours only) by adding a check to `PerfInstr::is_bypassible()` method.

> Problem could be with static cast. It add fffff at start.

Please share the code

pavelkryukov, 2018-03-26:
https://github.com/MIPT-ILab/mipt-mips/blob/f4f6cef1b2160312210cfce4c8cf904afcd831c8/simulator/bypass/data_bypass.h#L180 — that's the problematic line

Do you see failures in functional or performance mode?

inedostoev, 2018-03-26:
We all changed. But  this we couldn't understand. Before div worked correctly, in this case it doesn't work. Not correctly work mfhi, but why?

```bash
0x400638: divu $v0, $v1	 [ $hi = 0x0, $lo = 0x2 ]
0x40063c: mflo $a1	 [ $a1 = 0x2 ]
0x400640: addiu $at, $zero, 0x2	 [ $at = 0x2 ]
0x400644: bne $a1, $at, 1959
0x400648: mfhi $a1	 [ $a1 = 0x0 ]
0x40064c: bne $a1, $zero, 1957
0x400650: divu $v0, $v1	 [ $hi = 0x0, $lo = 0x2 ]
0x400654: mflo $a1	 [ $a1 = 0x2 ]
0x400658: addiu $at, $zero, 0x2	 [ $at = 0x2 ]
0x40065c: bne $a1, $at, 1953
0x400660: mfhi $a1	 [ $a1 = 0x0 ]
0x400664: bne $a1, $zero, 1951
0x400668: teq $a0, $zero
0x40066c: divu $v0, $a0	 [ $hi = 0x4, $lo = 0x0 ]
0x400670: mflo $a1	 [ $a1 = 0x0 ]
0x400674: bne $a1, $zero, 1947
0x400678: mfhi $a1	 [ $a1 = 0x0 ]
0x40067c: addiu $at, $zero, 0x4	 [ $at = 0x4 ]
0x400680: bne $a1, $at, 1944
0x4024e4: addiu $v0, $zero, 0x4	 [ $v0 = 0x4 ]
```

inedostoev, 2018-03-26:
And fall down func and perm, and rf tests

pavelkryukov, 2018-03-26:
Yes, please

pavelkryukov, 2018-03-26:
(adding 2 extra points due to increased complexity of division, multiplication, and data bypassing)

pavelkryukov, 2018-04-13:
@inedostoev @TimofeevAlex 

Can we proceed by starting with simple instructions only (no multiplication/division)?

inedostoev, 2018-04-13:
We have all instr, but now we write tests in tt.core

inedostoev, 2018-04-13:
We passed all test for 32-bit instrs

pavelkryukov, 2018-04-16:
OK. If 32 bit tests are working, please open a pull request so your changes are merged. That would shift the responsibility of merge from you to developers of new features.

pavelkryukov, 2018-05-04:
@TimofeevAlex @inedostoev 

That goes for a long time, whereas deadline is very soon.
I propose 3 options for you:
1. Preferrable option 
   * you merge your code to the latest master branch before the deadline
      * all *existing* tests are passed
      * I do not care whether doubleword instructions work or not
      * I do not care _much_ about quality of code (you may left sub-optimal solutions)
   * I give 4 points to each of you
   * I'll add all the tests 
2. Not realistic option, but I have to make an offer
   * you deliver everything before the deadline: instructions, tests, etc.
   * I give 6 points to each of you. I'm doing such a favor as you faced a lot of noise with multiplication/accumulation/unaligned memory — but it's a favor, as if you had delivered your patches before that, you would have not face those problems.
3. Dull option
   * you do nothing else
   * I give 2 points to each of you 
   * I merge your code on my own and complete all the stuff

You do not have to choose the option right now, the deadline for the task would be the deadline for your decision.

closed 2018-05-11 by pavelkryukov