MIPT-ILab/mipt-mips #221: Use «fast» LRU implementation for Cache class

pavelkryukov, 2017-11-13:
CacheTagArray implements LRU using a simple `std::list`. To perform a cache lookup, the list has to be fully iterated to find a position of a way, so lookup is O(N) operation. The time can be decreased if additional `std::unordered_map` is used like LRUCache class does

pavelkryukov, 2017-11-26:
However, if I were you I would continue with #77. It is better to have tests ready first, so you avoid making errors during refactoring and/or optimizations.

denislos, 2017-12-04:
When hash tables are used, miss_rate_sim takes 6 times less time.

pavelkryukov, 2017-12-05:
Nice, proceed with addition of hash table to LRUInfo class.

pavelkryukov, 2017-12-05:
You may speed up it more if you read memory access trace once (to some std::vector, for instance).

closed 2017-12-06 by pavelkryukov

pavelkryukov, 2017-12-06:
> You may speed up it more if you read memory access trace once (to some std::vector, for instance).

It is 5 seconds now. I must recall all the ILab diplomas from the guys who provided a cache implementation that ran for 30 minutes!