MIPT-ILab/mipt-mips #234: Implement software prefetching instructions

pavelkryukov, 2017-11-22:
MIPS IV introduces `pref` instruction which prefetches data from memory.

The 1st difficulty is encoding: `pref` instruction has a `hint` field which defines it behavior. We may think about prefetch instruction as a set of instructions (like REGIMM instructions):

Secondly, prefetching is not a data read: it generates no traps.

Finally, prefetching may have different outcome depending on hints

AndrewSultan, 2018-04-01:
What OperationType value should we use in pref instruction?

pavelkryukov, 2018-04-01:
You need to add a new type, as you have to set `dst = mips_zero` somewhere.

pavelkryukov, 2018-04-01:
Actually I advise you to start with #17.
It does not make sense to do prefetches if there is no data cache.