MIPT-ILab/mipt-mips #235: Add FPU coprocessor instructions to disassembler

pavelkryukov, 2017-11-22:
FPU instructions are part of MIPS ISA (see part B). We want to start with instruction decoding.

http://math-atlas.sourceforge.net/devel/assembly/mips-iv.pdf

pavelkryukov, 2019-03-18:
@DavidRabbitson 
The best way to generate unit tests is to:
1. Uncomment FPU instructions in https://github.com/MIPT-ILab/mips-traces/blob/master/tt.core.universal.s
2. Build the file with MIPS assembler
3. Dump its contents with MIPS objdump (`mips-unknown-gnu-linux-objdump -d filename.out`)

Then you can implement the code in TDD manner.

DavidRabbitson, 2019-03-19:
Ok. Now I'm learning FPU's instructions structure. First enhancements will be on these weekends.

DavidRabbitson, 2019-03-22:
How do I add instructions to `mips_instr.cpp` like `add.s` and `add.d` if they have the same function code?

pavelkryukov, 2019-03-22:
For example, you can add different table for different formats

DavidRabbitson, 2019-03-22:
How do I add tables to the decoder if instructions with `COP1=0x11` opcode have different defining fields?
https://github.com/MIPT-ILab/mipt-mips/blob/0d6542a52df388355eea50486ad81c8e0f971de8/simulator/mips/mips_instr.cpp#L378-L385
I mean instruction `add.s` is defined by the `funct` field and `bc1f` is defined by the `fmt`/`rs` field.

pavelkryukov, 2019-03-22:
I think it may be handled in a separate method where you can check functors etc.

```c++
 switch ( instr.opcode) 
 { 
     case 0x0:  return get_table_entry( isaMapR<I>,      instr.funct); 
     case 0x1:  return get_table_entry( isaMapRI<I>,     instr.rt); 
     case 0x10: return get_table_entry( isaMapCOP0<I>,   instr.rs);
     case 0x11: return get_cp1_entry( instr); 
     case 0x1C: return get_table_entry( isaMapMIPS32<I>, instr.funct); 
     default:   return get_table_entry( isaMapIJ<I>,     instr.opcode); 
 }
```

By the way, may you check-in the tests, even if they are disabled for now?

DavidRabbitson, 2019-03-22:
They do not pass because I didn't define `fp_*` type anywhere. Where can I define them?

DavidRabbitson, 2019-03-25:
How can I define the same instruction if in MIPS I it does not have any inputs except offset and in MIPS IV it has one more input?
[image]

pavelkryukov, 2019-03-25:
`cc = 0 implied` means that it is implied by compiler or assembler. We have to implement the generic case.

DavidRabbitson, 2019-03-25:
Instructions `mfc1` and `mtc1` have the same order of arguments `rt, fs`, but in first case rt is destination and in second is source. Can I reorder them like `mtc0` and `mfc0` or should I make them like reference says?
https://github.com/MIPT-ILab/mipt-mips/blob/ea14ec19a6a792905d4a57f87c9756a9093b3407/simulator/mips/t/unit_test.cpp#L2056
https://github.com/MIPT-ILab/mipt-mips/blob/ea14ec19a6a792905d4a57f87c9756a9093b3407/simulator/mips/t/unit_test.cpp#L2180

pavelkryukov, 2019-03-25:
I prefer to stick to our conventions where destination always goes first, and sources follow it.

pavelkryukov, 2019-03-25:
[ ] lwc1
 [ ] swc1
 [ ] ldc1
 [ ] sdc1
 [ ] lwxc1
 [ ] swxc1
 [ ] ldxc1
 [ ] sdxc1
 [ ] mtc1
 [ ] mfc1
 [ ] dmtc1
 [ ] dmfc1
 [ ] ctc1
 [ ] cfc1
 [ ] add
 [ ] sub
 [ ] mul
 [ ] div
 [ ] abs
 [ ] neg
 [ ] sqrt
 [ ] c
 [ ] recip
 [ ] rsqrt
 [ ] madd
 [ ] msub
 [ ] nmadd
 [ ] nmsub
 [ ] cvt
 [ ] round
 [ ] trunc
 [ ] ceil
 [ ] floor
 [ ] mov
 [ ] movt
 [ ] movf
 [ ] movz
 [ ] movn
 [ ] bc1t
 [ ] bc1f
 [ ] bc1tl
 [ ] bc1tf

DavidRabbitson, 2019-03-25:
Oh, thats better.

DavidRabbitson, 2019-03-26:
How can I add `Src3` to implement `madd.fmt` instructions? I don't feel that adding `src3` here is good idea.
https://github.com/MIPT-ILab/mipt-mips/blob/8d9569a5e84752fbd48ac9272b9e1b8ed79669db/simulator/mips/mips_instr.cpp#L228-L230

pavelkryukov, 2019-03-26:
Ouch. Let's put these instructions aside for now.

DavidRabbitson, 2019-03-27:
Do I need to implement 3rd source print?

pavelkryukov, 2019-03-27:
Yes, I think that would be useful

closed 2019-03-27 by pavelkryukov