MIPT-ILab/mipt-mips #277: Implement RISC-V RF

pavelkryukov, 2018-02-08:
RISC-V register file is very close to MIPS. It contains 32 registers which may contain 128, 64, or 32 bits of data. Register 0 is hardwired to zero signals.

https://en.wikichip.org/wiki/risc-v/registers

pavelkryukov, 2018-02-09:
@alex19999 We have a guideline ([pt. 7 of contributing guidelines](https://github.com/MIPT-ILab/mipt-mips/blob/master/CONTRIBUTING.md)) that students may not be assigned to 3 issues at once. Could you please proceed with your previous assignments before starting this?

alex19999, 2018-02-15:
Pavel Igorevich, as I understood, I can mainly copy mips_rf.h file and use RegNum class(because there are 32 registers two), but it's necessary to define the template of structure Reg, so it will be possible to create 32-, 64- and 128-bits registers. Can you give some additional information? Thank you.

pavelkryukov, 2018-02-15:
Yes, that is ok. Please note that I have added HI and LO registers which are MIPS-specific.

pavelkryukov, 2018-03-01:
@alex19999 Do you have any work in progress on the issue? I'm going to do some code rearrangements but I don't want to interfere with you here.

alex19999, 2018-03-01:
Pavel Igorevich, hello. I started this task, but did nearly nothing. Could you advise me what I have to do firstly: Instr Cache task or this? I hope I'll do it tomorrow. I need to do some other tasks today01.03.2018, 20:27, "Pavel I.

pavelkryukov, 2018-03-01:
OK.
Proceed with instruction cache. I'll update RF task by EOW.

alex19999, 2018-03-17:
Pavel Igorevich, hello. Could you tell me, please, what variant is better?
Now in rf.h we have something like that:
`  struct Reg { 
  uint32 value = 0;
   };`

I think, that it is possible to do like that
`template <typename ISA, typename T>`
 `class RF`
 `  struct Reg { 
  T value = 0;
   };`

or may be it's better to use union instead of the structure. I mean:
`union Reg {
uint32;
uint64;
uint128;
}`

alex19999, 2018-03-17:
or even separate class inside RF
`template <typename T>`
`class Register`

pavelkryukov, 2018-03-18:
I would put RegisterUInt into ISA class:

```c++
template <typename ISA>
class RF {
    using RegisterUInt = typename ISA::RegisterUInt;
    struct Register {
        ISA::RegisterInt value = 0u;
     };
```
 
Then 3 ISA classes may be defined: RISC32, RISC64, and RISC128, varying RegisterUInt type

alex19999, 2018-03-18:
Pavel Igorevich, you've said (our conversation about MIPSRegister tests):

> 1. I expect that no register is HI, LO or HI_LO

> 2. I expect that registers generated by mips_hi, mips_lo, and mips_hi_lo are invalid and their usage will lead to error
etc.

Now is it possible to do this tests printing out names of the registers? Is it not very silly in this case?)

alex19999, 2018-03-18:
I decided that it's better to add some strings in c-tor:
```
explicit RISCVRegister( RegNum id) : value( id) 
{
	if( id == MIPSRegister::mips_hi.to_size_t() ||
	    id == MIPSRegister::mips_lo.to_size_t() ||
	    id == MIPSRegister::mips_hi_lo.to_size_t()) 
	{
		std::cerr << "ERROR: Try to generate RISCV register by -hi, -lo or -hi_lo register = " << id;
		exit( EXIT_FAILURE);
	}
}
```
And in the unit_test I'll add necessary checks

pavelkryukov, 2018-03-19:
No, linking RISCV implementation to MIPS implementation does not look good.

You may initialize `RISCVRegister::mips_hi` with MAX_REG and add a check into ` RISCVRegister::to_size_t` that current value is not equal to it.

pavelkryukov, 2018-03-19:
To continue, I advise you to instantiate these classes in perf_sim.cpp:

```c++
#include "risc_v/risc_v.h"

template<> RF<RISCV>; // step 1
template<> DataBypass<RISCV>; // step 2
template<> InstrMemory<RISCV>; // step 3
template<> FuncSim<RISCV>; // step 4
template<> PerfInst<RISCV>; // step 5
template<> Fetch<RISCV>; // step 6
template<> Writeback<RISCV>; // step 7
```

and create all dummy interfaces for RISCVInstr to build the code.

alex19999, 2018-03-19:
Ok, I'll do it. Thank you for comments

pavelkryukov, 2018-03-19:
Sorry, I forgot the syntax for template instantiation. It should be:

```c++
#include "risc_v/risc_v.h"

template class RF<RISCV>; // step 1
template class DataBypass<RISCV>; // step 2
template class InstrMemory<RISCV>; // step 3
template class FuncSim<RISCV>; // step 4
template class PerfInst<RISCV>; // step 5
template class Fetch<RISCV>; // step 6
template class Writeback<RISCV>; // step 7
```

alex19999, 2018-03-20:
Pavel Igorevich, may I do like this:
```
template <typename T>
struct RISCV {
    using RegisterUINT = T;
```
So in perf_sim.cpp will be:
```
template class RF<RISCV<uint32>>;
```

pavelkryukov, 2018-03-20:
Yes, sure. You may add aliases too:

```c++
using RISCV32 = RISCV<uint32>;
using RISCV64 = RISCV<uint64>
using RISCV128 = RISCV<uint128>
```

alex19999, 2018-03-21:
Pavel Igorevich, sorry, but I am need to ask you again. I  instantiated  all(besides `FuncSim<RISCV>`) classes and added all necessary methods in `riscv_instr.h`(so, we can build the project). So, the problem I have is connected with `FuncSIm<RISCV>`. When I try to build I get such things `undefined reference to FuncSim<RISCV<unsigned int> >::~FuncSim() `, but I have all these functions( I didn't change anything in this file) and all needed libraries are included( I hope). 
 In addition, I changed ` const union _instr` using this manual https://riscv.org/specifications/ . For example, I defined B format like this:
```
const struct AsB
            {
                uint32 imm4   :1;
                uint32 imm3   :6;
                uint32 rt     :5;
                uint32 rs     :5;
                uint32 funct3 :3;
                uint32 imm2   :4;
                uint32 imm1   :1;
                uint32 opcode :7;
            } asB;
```
Thus, I need your help with FuncSim and I am going to change `enum OperationType`, then I will create a pull request. Thank you

pavelkryukov, 2018-03-21:
Do you have these in `func_sim.cpp`:

```c++
template class FuncSim<RISCV32>;
template class FuncSim<RISCV64>;
template class FuncSim<RISCV128>;
```

pavelkryukov, 2018-03-21:
> I changed const union _instr using this manual

Are you implementing internals of RISC-V class? It is a separate task we don't want to do yet. The right way is to use RISC-V open source decoder to get as much code generated as possible.

alex19999, 2018-03-21:
Oh, I forgot to add these strings in these file, thank you. No, I just had defined the `riscv_instr.h` file and then consistently instantiated all necessary classes. During this, I added the necessary methods and variables in `RISCVInstr` mainly copying `MIPSInstr`. But I understand that `RISCVInstr` has some differences. As the result, I opened manual and try to fix just `const union instr` and `enum OperationType`. That's all. Open source decoder?

pavelkryukov, 2018-03-21:
You may remove unions and OperationType for now as they are invisible externally. 

RISC-V community has a tool to generate ISA in several formats: https://github.com/riscv/riscv-opcodes, so if the committee defines new instruction, we will get them soon.

alex19999, 2018-03-21:
But, for example, I need function `bool has_trap() const { return trap != TrapType::NO_TRAP; }` somewhere. So, is it possible just to do like this ` `bool has_trap() const { return false}`(without defining TrapType) ?

pavelkryukov, 2018-03-21:
Yep. We will define them later, so RISC-V implementation may be performed step-by-step.

alex19999, 2018-03-21:
Ok, sorry for my haste

commit f4f6cef1 (2018-03-22): RISC-V Instruction interface (fixes #277)
closed 2018-03-22 by pavelkryukov