MIPT-ILab/mipt-mips #282: Add MIPS32 accumulating instructions

pavelkryukov, 2018-02-10:
MIPS32 introduces 4 accumulation instructions: `madd`, `maddu`, `msub`, `msubu`. Unlike common instructions, they don't overwrite the contents of HI/LO register pair, but accumulate results:
```c++
    hilo += rs * rt;
```

The roadmap for implementation:
 * [ ] Add accumulation feature to register file
 * [ ] Introduce instruction to MIPS decoding flow
 * [ ] Add a flag to perform accumulation instead of writing

pavelkryukov, 2018-03-12:
@AndrewSultan 

The idea of this instructions is to have an additional ALU which operates over HI/LO registers. To model it, you have to introduce new interfaces to RF class which does addition or subtraction instead of simple overwrite.

Additionally, such instructions cannot deliver data to bypass logic. Please consult with @denislos to understand the best way to change model properly.

AndrewSultan, 2018-03-26:
For an unknown reason the madd test failed, but maddu test passed.

pavelkryukov, 2018-03-26:
https://github.com/MIPT-ILab/mipt-mips/blob/355fec18c1bcb8ccb53d718791c51161b8ed53f3/simulator/func_sim/rf/rf.h#L47

That looks suspicious, as all operations are unsigned. However I don't see an obvious bug...

What are the operands for failing test, actual output, and expected output?

AndrewSultan, 2018-03-26:
0x400b54: madd $a0, $a0  [ $hi = 0x1, $lo = 0x0 ]
0x400b58: mfhi $v1       [ $v1 = 0x2 ]
_____________________________________________________________
Actual msg:
[  DEATH   ] Mismatch: 
[  DEATH   ] Checker output: 0x400b58: mfhi $v1	 [ $v1 = 0x2 ]
[  DEATH   ] PerfSim output: 0x400b58: mfhi $v1	 [ $v1 = 0x1 ]
[  DEATH   ] 
[  FAILED  ] Perf_Sim.Run_Full_Trace (14 ms)
[ RUN      ] Perf_Sim.Run_SMC_Trace
[       OK ] Perf_Sim.Run_SMC_Trace (5 ms)
[----------] 2 tests from Perf_Sim (19 ms total)

pavelkryukov, 2018-03-26:
Double-check that condition:

https://github.com/MIPT-ILab/mipt-mips/blob/355fec18c1bcb8ccb53d718791c51161b8ed53f3/simulator/core/perf_instr.h#L24

pavelkryukov, 2018-03-27:
Denis @denislos, could you please take a look? It seems that MADD bypasses `0x1` to HI register, while it should be prevented by `is_bypassible()` check.

pavelkryukov, 2018-03-27:
https://github.com/MIPT-ILab/mipt-mips/blob/48860703d0a002e94a64752dd9f51228991f02db/simulator/mips/mips_instr.h#L259

BTW, you don't need these functions as they do the same thing as `execute_mult` and `execute_mult`; all special behavior is controlled by operation type

denislos, 2018-03-27:
There is a strange thing, I have run the tests and got this

fetch   cycle 2491: 0x400b3c: 0x0x400b3c: mfhi $v1
decode  cycle 2491: 0x400b38: madd $a0, $a0
execute cycle 2491: 0x400b34: lui $a0, 0x1	 [ $a0 = 0x10000 ]
memory  cycle 2491: 0x400b30: mthi $zero	 [ $hi = 0x0 ]
wb      cycle 2492: 0x400b30: mthi $zero	 [ $hi = 0x0 ]
Executed instructions: 944

decode  cycle 2492: 0x400b3c: mfhi $v1 (data hazard)
execute cycle 2492: 0x400b38: madd $a0, $a0	 [ $hi = 0x1, $lo = 0x0 ]
memory  cycle 2492: 0x400b34: lui $a0, 0x1	 [ $a0 = 0x10000 ]
wb      cycle 2493: 0x400b34: lui $a0, 0x1	 [ $a0 = 0x10000 ]
Executed instructions: 945

decode  cycle 2493: 0x400b3c: mfhi $v1 (data hazard)
execute cycle 2493: bubble
memory  cycle 2493: 0x400b38: madd $a0, $a0	 [ $hi = 0x1, $lo = 0x0 ]
In DECODE_2_FETCH_STALL port data was added at 2492 clock and will not be readed

It looks like that  the fetch stage does not handle stalls in case of a cache miss. I have changed `fetch` a little bit in order to fix it
template <typename ISA>
void Fetch<ISA>::ignore( Cycle cycle)
{
    /* ignore PC from other ports in the case of cache miss */ 
    rp_external_target->ignore( cycle);
    rp_hold_pc->ignore( cycle);
    rp_target->ignore( cycle);
    `rp_stall->ignore( cycle);`

    /* ignore miss signal */
    rp_hit_or_miss->ignore( cycle);
}

Now it passes all the tests.

pavelkryukov, 2018-03-27:
Thank you. There is no problem with bypass, right?

pavelkryukov, 2018-03-27:
> In DECODE_2_FETCH_STALL port data was added at 2492 clock and will not be readed`

@AndrewSultan Do you have the same message as Denis has?

pavelkryukov, 2018-03-27:
Please proceed by updating submodule and implementing subtracting instructions. I think the best strategy is to return `int8` for `is_accumulating()`: -1 does subtraction, 1 does addition, 0 does nothing.

commit 4903628a (2018-03-28): Add msub and msubu instructions (fixes #282)
closed 2018-03-28 by pavelkryukov