MIPT-ILab/mipt-mips #290: Implement interfaces to GDB

pavelkryukov, 2018-02-21:
GDB may use external simulators for debug. To do this, several interfaces should be implemented: https://www.embecosm.com/appnotes/ean3/embecosm-howto-gdb-porting-ean3-issue-2.html#ref_simulators

These interfaces should be implemented in FuncSim class and wrapped around with C functions in separate C file. New target libsim.a should be added to Makefile.

pavelkryukov, 2018-03-01:
@trexxet Do you want to participate?

pavelkryukov, 2018-07-19:
An article about integration of GDB into cycle-accurate simulation: https://www.embecosm.com/appnotes/ean7/embecosm-gdb-verilator-ean7-issue-1.html

pavelkryukov, 2018-07-23:
@trexxet Do you have any progress on that? If you have issues, I advise to put other tasks to higher priority, as they are simpler IMO.

trexxet, 2018-07-23:
@pavelkryukov Yes, I'm still working on it. There are some issues which I'm able to overcome, however, I may switch to other tasks.

pavelkryukov, 2018-07-23:
I suggest you to make a pull request so I may see the architecture of your code, do advises, and think about the issues. Probably, some code may be pushed at the moment, like clean-ups or additional interfaces of the simulation core.

trexxet, 2018-07-30:
As I figured out, GDB can't just connect to simulator compiled into libsim.a. 
GDB sources must be reconfigured to use the external simulator (by default, GDB doesn't even has `target sim`) and rebuilt; then libsim.a is linked during build time. This requires some workarounds with autotools configs. I think that creating patch files for them fits well.

Moreover, some headers from GDB sources are required for wrappers. I suggest adding `binutils-gdb` as  a git submodule as a robust solution.

pavelkryukov, 2018-07-30:
> I suggest adding binutils-gdb as a git submodule as a robust solution.

I have concerns over adding a sub-module to main repository once again, as it creates much confusion for project newcomers. They do not have to suffer from GDB integration unless they are using it or develop it.

Additionally, I would not call it robust. If you are using Git as source control for binutils-gdb, you have to fetch all the bugs it has at the moment. Fetching gzipped sources of exact version is more reliable, but it is not a choice since we have to do patches.

> This requires some workarounds with autotools configs. I think that creating patch files for them fits well.

Maybe the better thing would be to do a fork of GNU Binutils repository to have all the patches inside. Then, if someone needs to build GDB integrated with MIPT-MIPS, he or she should clone both repositories and run some script or Makefile.

pavelkryukov, 2018-07-30:
> Maybe the better thing would be to do a fork of GNU Binutils repository to have all the patches inside. 

And make MIPT-MIPS a submodule inside it!

pavelkryukov, 2018-07-31:
https://github.com/MIPT-ILab/mipt-sim-gdb – here you go. You should have full access to write and push.

pavelkryukov, 2018-08-22:
Two weeks left before the deadline.
With given tempo, the issue is not expected to be fixed before deadline.

trexxet, 2018-09-04:
I'm still working on it. Writing interfaces is not hard, the problem is to build simulator with some GDB source files, which are configured by GNU Autotools and depend on external libraries (e.g. `binutils-dev`), via CMake. However, I'll accomplish it soon.

pavelkryukov, 2018-09-04:
> e.g. binutils-dev

Could you please explain more details? I used to build GDB from scratch and usually it has less to no requirements

trexxet, 2018-09-04:
Standalone GDB builds finely. However, GDB passes program to simulator via BFD (Binary file descriptor), which is part of binutils.

pavelkryukov, 2018-09-04:
> I used to build GDB from scratch and usually it has less to no requirements

Same worked for all the GNU Binutils.

pavelkryukov, 2018-09-04:
> is to build simulator with some GDB source files, which are configured by GNU Autotools and depend on external packages (e.g. binutils-dev), via CMake. 

What is

pavelkryukov, 2018-09-04:
Oh I see, you have issues as auto-generated header file should be included to MIPT-MIPS translation unit which defines the extrenally linked interfaces, right?

trexxet, 2018-09-06:
Yes, however, I did a workaround and wrote that files (or, more correctly, stubs) by myself. 

At this point, I have successfully built GDB with MIPT-MIPS simulator, and connected to simulator from GDB. I'm going to commit it as soon as I'll test the current interfaces (creating and closing simulator). It remains for me to create other interfaces (step-by-step execution, printing registers etc.), however, it won't be hard. I think the most complicated part is done.

trexxet, 2018-09-06:
I'll make them right after interfaces: running test requires an ELF file (I've used one in `traces/`) and communicating with GDB.

I've pushed current state to `gdb_interface` in my fork (34e492f2238b330338356e34e30821b7f2e484a9).

To build:
1) In GDB source directory, run `./configure --target=mips`
2) Then, build libsim.a from simulator submodule (target `mipt-mips-gdb`)
3) Build GDB: `make`

To run: `cd gdb`, then:
`./gdb /path/to/binary`
`target sim`
`load`
`run`
or:
`./gdb`
`target sim -b /path/to/binary`
`load`
`run`

pavelkryukov, 2018-09-06:
ELF binaries are built by CI. Can communicating with GDB be modeled? These are unit tests, and the idea is to decouple them out of outer environment

pavelkryukov, 2018-09-06:
Please put build and run instruction to a new Wiki page

pavelkryukov, 2018-09-07:
> run `./configure --target=mips`

Should it be "mipsel"? We have only little-endian MIPS at the moment.

trexxet, 2018-09-07:
> Should it be "mipsel"?

GDB seems not to have target "mipsel" ([here](https://github.com/bminor/binutils-gdb/blob/master/gdb/configure.tgt)). Moreover, I removed all targets except "mips", however, autoconf sets target as host, if `--target` not specified.

> Can communicating with GDB be modeled? 

We can emulate GDB by calling interface functions. I'll do unit test right after finishing interfaces.

I've pushed implementation for memory access (b11d0cf5456fde2e04d8e100f1fba9b7054adcd7).

pavelkryukov, 2018-11-22:
@trexxet Do you have next steps?

pavelkryukov, 2018-12-15:
Completing this manually, since the original owner is longer responsive

closed 2018-12-16 by pavelkryukov