MIPT-ILab/mipt-mips #301: Add DataBypass description to Wiki

pavelkryukov, 2018-02-28:
Please, explain behavior of DataBypass class on our Wiki.
Use this page as an example: https://github.com/MIPT-ILab/mipt-mips/wiki/BPU-model

Please try to avoid describing C++ internals and using C++ code, excepting interfaces, Try to focus on explanation "How it works? How are special cases handled? What has to be done to extend functionality?"

pavelkryukov, 2018-03-23:
Link to the page: https://github.com/MIPT-ILab/mipt-mips/wiki/Data-Bypass-and-Scoreboard

pavelkryukov, 2018-06-25:
@denislos I still expect the manual...

denislos, 2018-07-02:
I am sorry. I had a cold last week, so I am planning to complete it soon by updating the manual with a solution of #424.

pavelkryukov, 2018-07-26:
@denislos We still this manual to explain #463 and #17 more explicitly.

pavelkryukov, 2018-08-27:
@denislos It's quite late to have it for summer tasks, but in October that would be actual for me again.

pavelkryukov, 2018-10-01:
Oh, that's October.

denislos, 2018-10-03:
It will be finished by the end of the week.

pavelkryukov, 2018-11-25:
@denislos Just a reminder...