MIPT-ILab/mipt-mips #303: Update information about disassembly

pavelkryukov, 2018-02-28:
Hi @TimofeevAlex 

Previously you did a lot of work on MIPS instructions decoder.
Could you please spend some time to update this wiki page a little? 
https://github.com/MIPT-ILab/mipt-mips/wiki/MIPS-disassembler

Thanks

TimofeevAlex, 2018-02-28:
Yes, I'll be glad to do it.

Assigned #303 <https://github.com/MIPT-ILab/mipt-mips/issues/303> to
@TimofeevAlex <https://github.com/timofeevalex>.

<https://github.com/MIPT-ILab/mipt-mips/issues/303#event-1497499864>, or mute
the thread

pavelkryukov, 2018-10-10:
@alexunderch It might be not clear what is requested, but the general idea is to update referenced Wiki manual with description of the _actual_ code of MIPS instruction decoder.

alexunderch, 2018-10-10:
That was clear for me. Thanks.

Sincerely,
Alexander Chernyavskiy.