MIPT-ILab/mipt-mips #365: Fix bypassible condition

pavelkryukov, 2018-03-27:
https://github.com/MIPT-ILab/mipt-mips/blob/355fec18c1bcb8ccb53d718791c51161b8ed53f3/simulator/core/perf_instr.h#L24

This condition returns true for accumulating instructions, potentially leading to failures.

AndrewSultan, 2018-03-27:
`auto` is_bypassible() const { return !this->is_conditional_move() || !this->is_accumulating_instr(); } 
I added a negation but the error was not eliminated

pavelkryukov, 2018-03-27:
```c++
auto is_bypassible() const { return !this->is_conditional_move() || !this->is_accumulating_instr(); }
```

Your condition always returns true now.

AndrewSultan, 2018-03-27:
Oh, I realized my mistake

AndrewSultan, 2018-03-27:
`auto is_bypassible() const { return !this->is_conditional_move() && !this->is_accumulating_instr(); }`
maybe so?

pavelkryukov, 2018-03-27:
At least this does not break current simulation, please open a pull request

commit 48860703 (2018-03-27): Fix is_bypassible condition for accumulating instructions (fixes #365)
closed 2018-03-27 by pavelkryukov