MIPT-ILab/mipt-mips #451: Enhance cache line size study

pavelkryukov, 2018-07-02:
@alex19999: see #18

alex19999, 2018-07-06:
Sorry for delay, I completely forgot about this task and didn't check my mail. I have some questions and assumptions. First of all, where can I use pipeline depth in this study? Is it connected with time access in case of a miss( miss penalty)? Also I can't find a place for exponent in this formula. Is it connected with some probabilistic considerations?

pavelkryukov, 2018-07-06:
I expect you to answer these questions :-)
There is a simple arithmetic to get IPC from basic parameters, as I see. It's simpler to start with CPI though.
Additionally, everything is fully determenistic

pavelkryukov, 2018-08-27:
So what happened to this? Should I do it myself before new year starts?