MIPT-ILab/mipt-mips #461: Add option to set writeback bandwidth

pavelkryukov, 2018-07-08:
Blocked by #383

pavelkryukov, 2018-07-25:
@YanLo You may be also interesting in closing this, as it is the motivation for adding sequence identifiers to instruction class. We want to be able to receive two instructions to writeback stage, and perform writebacks in program order to maintain architecture state.

DavidRabbitson, 2019-03-28:
Am I right that I need to rework this method so it'll have additional `bandwidth` argument?
https://github.com/MIPT-ILab/mipt-mips/blob/79120212ded48f95902a449c61a0e505b3e67e5c/simulator/modules/writeback/writeback.h#L75

pavelkryukov, 2019-03-28:
That may be one of the possible implementations.
Alternatively, you may add a "set_bandwidth" method,

DavidRabbitson, 2019-04-01:
I'm not sure what bandwidth does. As I can see here, bandwidth argument is `PORT_BW = 1`. So, I replaced this constant with `uint32 bandwidth` that I implemented as a default argument in `writeback` constructor.
https://github.com/MIPT-ILab/mipt-mips/blob/d8966dd60015cfd2dada9955eccf8c38a775529e/simulator/modules/writeback/writeback.cpp#L13-L14
I'm not sure if my changes were appropriate and I did something right. What should I rework?

pavelkryukov, 2019-04-01:
Currently WB stage can handle only a single instruction, dispatched either from simple EXE module, MEM module, or BRANCH module, The objective is to allow handle 2 or more instructions coming from different sources.

For example, `fib.out` trace has a bubble in execution due to conflict between branches and additions for writeback stage. In target implementation, there should be no conflict.

DavidRabbitson, 2019-04-01:
Do I have to handle bypass and halt bandwidth or it should stay `PORT_BW`?

pavelkryukov, 2019-04-01:
Please figure out whether they are related to the issue.
I suggest you to draw a full topology of modules on the paper (what modules are connected with which ports)

pavelkryukov, 2019-04-15:
Let's draw a diagram of `fib.out`. It contains two instructions: jump and add.
However, they cannot be executed consequently, as it would require TWO instructions to do writeback. One instruction comes from Branch stage (B) and the other one comes from Execution stage directly (E):

```
j loop              | F D E B W
add $t3, $t2, $t1   |   F D * E W 
move $t1, $t2       |     F * D E W
move $t2, $t3       |         F D E W
```

If Writeback stage had more bandwidth, we would support this behavior:

```
j loop              | F D E B W
add $t3, $t2, $t1   |   F D E W 
move $t1, $t2       |     F D E W
move $t2, $t3       |       F D E W
```

DavidRabbitson, 2019-04-15:
That explains a lot, thanks :)

DavidRabbitson, 2019-04-17:
How do I deal with stall (data hazard on `add`) that `j` induce? May I rewrite decode module?

pavelkryukov, 2019-04-17:
Do you know what condition induces the stall? Can it be tuned to support variable bandwidth?

DavidRabbitson, 2019-04-18:
I figured that out. It can be by adding writeback_bandwidth to the FuncUnitInfo:
https://github.com/MIPT-ILab/mipt-mips/blob/3e500b950d4dad66a789a9cfe593646f1fd23726/simulator/modules/decode/bypass/data_bypass.h#L104-L107
and changing this return depending on this new variable:
https://github.com/MIPT-ILab/mipt-mips/blob/3e500b950d4dad66a789a9cfe593646f1fd23726/simulator/modules/decode/bypass/data_bypass.h#L47-L49
Now I have problem with ports. WB is almost working, but 2nd instruction that comes after add (`move t2, t3`) can't find t3 value so it becomes `0xdeadbeef`. I guess I write `t3` to the bypass port wrong. So, how do I read from port with != 1  bandwidth? Also, how do I write to ports with != 1 bandwidth?

pavelkryukov, 2019-04-18:
Ports have a bandwidth parameter (usually PORT_BW)

DavidRabbitson, 2019-04-18:
Well, now it does work, BUT there is no logic in execute stage right now. I mean if we have `writebeck bandwidth = 2` we have to choose which WB contains right value for the next instruction we might bypass. In current state, we always choose second (or the latest) instruction to get WB value from for the bypass. How can I optimise it?

pavelkryukov, 2019-04-18:
Okay, that goes beyond the scope of the task.
Is that mode (always choose second) functionally correct?

DavidRabbitson, 2019-04-18:
Well, fib.out works with 17+ cycles.

DavidRabbitson, 2019-04-18:
Referenced code is not ready yet, I have to connect bypass's writeback_stage_info structure so it might be adjusted from the perf_sim as well. Furthermore, there is some code to be improved.

commit d0a17020 (2019-04-20): Add writeback bandwidth set option (#959)

Fixes #461
closed 2019-04-20 by pavelkryukov