MIPT-ILab/mipt-mips #465: Remove type-punning from MIPS instruction

pavelkryukov, 2018-07-09:
It is still unclear to me whether type punning over POD types via union is legal or not in C++. So let's follow code guidelines and remove union, using hierarchic functions instead:

```c++
   struct bytes_ {
      struct AsR {
          int opcode : 5
// ...
     };
     AsR get_as_r() const {
         AsR result;
         result.opcode = (bytes & OPCODE_MASK) >> OPCODE_SHIFT;
     };
     uint32 bytes;
```

Additionally, it should simplify implementation of #387.

pavelkryukov, 2018-10-16:
Do you have any progress in these two weeks?

agrachiv99, 2018-10-17:
> Do you have any progress in these two weeks?

Sorry, I had some problems with setting up into the new project. I guess, I will finish with both of my issues during this week.

pavelkryukov, 2018-10-24:
One more week passed...

commit c5ff18a7 (2018-11-16): Remove type-punning from MIPS instruction

Fixes #465
closed 2018-11-16 by pavelkryukov