MIPT-ILab/mipt-mips #493: Add wiki page about factory patterns

pavelkryukov, 2018-07-22:
We use factory patterns for two classes: BP and Simulator, and we are going to spread the approach to new classes as well. Because of that we want to have the pattern described in our Wiki.

The structure of the page should follow the agenda:
1. Motivation — why do we need factories?
2. Step-by-step building: interface class, polymorphic derivatives, using `std::map` etc.
3. Overheads and issues.
4. Further reading (usually refers to books/articles author read to prepare the page).

I advise to look through "Gang of Four" book (a.k.a. _Design Patterns: Elements of Reusable Object-Oriented Software_, please demand it from the mentor), Wikipedia article, and our source code database (bpu.cpp and simulator.cpp).

davtyan-arsen, 2018-11-26:
So, could I get a copy of the book somehow?

pavelkryukov, 2018-11-26:
https://lib.mipt.ru/book/25344