MIPT-ILab/mipt-mips #512: Add unit tests for writeback module

pavelkryukov, 2018-07-26:
Writeback module is the simplest module to test. Its inputs are completed instructions, and its outputs are very few.

The idea is to wrap WB module inside a single environment with functional simulation:
```c++
class WritebackTesting {
    Writeback wb;
    FuncSim generator;
    /* ports to match WB ports */
};

graudtV, 2020-12-02:
Hi. I've written some facilities for testing, what is the best way to share this code with you? It is quite big to paste it here, but not finished yet to make a pull request.

pavelkryukov, 2020-12-02:
Thanks. I see it in a fork repo and can comment.

pavelkryukov, 2020-12-04:
We finalizing rating on 07.12.2020.
In order to get points for this, please complete this before this Sunday.

graudtV, 2020-12-04:
Ok. Could you please have a look? If it's ok, I will add some more tests

pavelkryukov, 2020-12-04:
If everything looks, please submit pull request to see all changes at once.

graudtV, 2020-12-06:
I am trying to write tests for predicted and not predicted instructions, and I a bit don't understand how Branch::is_misprediction() works.
Let we have an instruction, which is predicted as taken and is actually taken. This **is not** a misprediction. Condition on line 51 returns false (it is in BP). Condition on line 57 also returns false, because actual information is the same as in bp_data. The last expression returns true, because target != new_PC. So functions says it **is** misprediction.
https://github.com/MIPT-ILab/mipt-mips/blob/a39e74dd39b44c75d1bb76593085f6889b7a0dea/simulator/modules/branch/branch.h#L49-L61

pavelkryukov, 2020-12-06:
>  which is predicted as taken and is actually taken. This is not a misprediction. 
> because target != new_PC. So functions says it is misprediction

Correct: you predicted direction, but not the target (i.e. taken, but somewhere else).
That may happen in case of aliasing in branch prediction table.

pavelkryukov, 2020-12-06:
It is a great achievement for our project, as your infrastructure has high quality and may be easily extended for more modules.
Thank you, and keep that level in future work!

closed 2020-12-06 by pavelkryukov