MIPT-ILab/mipt-mips #520: Implement topology graph visualizer tool

pavelkryukov, 2018-07-27:
With #357, we may dump topology information:
* modules
* ports connecting modules
* ports fanouts
* ports BW
* ports latencies

Information should be dumped into `topology.json` file if a special command line option is provided. Then, stand-alone visualizing tool may fetch the data and do the work.

That information may be visualized in a pretty graph. Feel free to use existing tools in JavaScript, Python, or any other language or framework.

pavelkryukov, 2019-04-17:
@agrachiv99 That one is also about GUI, but I don't feel it is bold for informatic course.

exucutional, 2019-10-31:
Do I just need to generate svg/png/jpg picture? Or should I make interactive GUI with clickable modules and ports?

pavelkryukov, 2019-10-31:
> Do I just need to generate svg/png/jpg picture? Or should I make interactive GUI with clickable modules and ports?

What are the requirements for each of the solutions?

pavelkryukov, 2019-10-31:
Could you please share your plan for this work? What is going to be delivered first: JSON interface for simulator or the tool itself?

For option 1, please start with unit tests for Module/Root/Port class
For option 2, please first deliver a dummy JSON file which you are going to visualize.

exucutional, 2019-10-31:
I have already implement dummy JSON file with c3d029632203a2a085354b9905aae7cab5600b6e.
Example: 

       [
          {
            "module_name": "fetch",
            "write_ports": [
                {
                    "key": "FETCH_2_DECODE",
                    "bandwidth": "1"
                },
                ...
              ]
              "read_ports": [
                {
                    "key": "DECODE_2_FETCH_STALL",
                    "latency": "1"
                },
                ...
              ]
           }
           ...
        ]

About tool. I want use [SchemDraw ](https://cdelker.bitbucket.io/SchemDraw/). I just transform json into suitable form and generate picture. Also I found example of [interactive flowchart ](https://gojs.net/latest/samples/dynamicPorts.html)(unfortunately GoJS isn't free). Something like this is more complicated and I don't know yet how long it will take.

pavelkryukov, 2019-10-31:
> I want use SchemDraw . 

The license is compatible, but I'm afraid it does not satisfy our needs.
We have 3 types of relations:
1. Module can be a submodule of other module — the best option is to draw a square inside a square.
2. Module may have a read port — should look like a line out of the square
3. Module may have a write port — should look like a line with arrow

[image]

Electricity circuits may render only the second type of relations.

pavelkryukov, 2019-10-31:
> I have already implement dummy JSON file with c3d0296.

Please deliver unit tests for that implementation. How can we be sure that code is working?

Also, we have all the infrastructure for test-driven development  to produce reliable code faster.
I suggest to check our [Wiki lesson](https://github.com/MIPT-ILab/mipt-mips/wiki/Introduction-to-Test-Driven-Development).
Then you may open a pull request for a code review.

pavelkryukov, 2019-11-23:
Could you please update on this?

exucutional, 2019-11-24:
I am in process. Current interactive demo looks like this
[image]

exucutional, 2019-11-24:
Later I will fix labels and connections overlap to better generate module positions. Also I want display port info on hover.

pavelkryukov, 2019-11-24:
Looks impressive!

I suggest to move port name on hover, but use different line weights depending of size of token.
Additionally, I think we may exclude looping ports, they create a lot of noise but does not bring much information.

pavelkryukov, 2019-11-27:
I propose to start code review procedure, as the fixes have minor scope and may be applied later.

pavelkryukov, 2019-11-29:
I got this:

[image]

pavelkryukov, 2019-11-29:
OK, it requires the full path. But it's better to remove the hardcoded location, mostly probably there will no such file.

pavelkryukov, 2019-11-29:
https://github.com/MIPT-ILab/mipt-mips/blob/master/visualizer/README.md — could you please provide a description here and how the tool is connected to MIPT-V (also, please add a dash to the name)

PR #1191 (2019-12-02): Add config classes in visualizer
Fixes #1185 
Fixes #520
closed 2019-12-02 by pavelkryukov