MIPT-ILab/mipt-mips #563: Uninitialized value in LRUCache

pavelkryukov, 2018-09-11:
From Clang-Tidy report:

```c++
/home/travis/build/MIPT-ILab/mipt-mips/simulator/infra/instrcache/t/../LRUCache.h:28:29:warning:Undefined or garbage value returned to caller [clang-analyzer-core.uninitialized.UndefReturn]
        auto size() const { return number_of_elements; }
                            ^
/home/travis/build/MIPT-ILab/mipt-mips/simulator/infra/instrcache/t/unit_test.cpp:116:12:note:Calling 'LRUCache::empty'
    CHECK( instr_cache.empty());
           ^
/home/travis/build/MIPT-ILab/mipt-mips/simulator/../external/catch.hpp:13093:97:note:expanded from macro 'CHECK'
#define CHECK( ... ) INTERNAL_CATCH_TEST( "CHECK", Catch::ResultDisposition::ContinueOnFailure, __VA_ARGS__ )
                                                                                                ^
/home/travis/build/MIPT-ILab/mipt-mips/simulator/../external/catch.hpp:1708:70:note:expanded from macro 'INTERNAL_CATCH_TEST'
            catchAssertionHandler.handleExpr( Catch::Decomposer() <= __VA_ARGS__ ); \
                                                                     ^
/home/travis/build/MIPT-ILab/mipt-mips/simulator/infra/instrcache/t/../LRUCache.h:29:37:note:Calling 'LRUCache::size'
        bool empty() const { return size() == 0; }
                                    ^
/home/travis/build/MIPT-ILab/mipt-mips/simulator/infra/instrcache/t/../LRUCache.h:28:29:note:Undefined or garbage value returned to caller
        auto size() const { return number_of_elements; }
```

agrachiv99, 2018-10-06:
I have several problems with compilation. 
 
1) In "readme" file it's written that i have to "Go to traces subdirectory and build MIPS traces just by typing make", but master brunch doesn't contain traces at all. So should i clone that tracers brunch in order to build the project?

2) Some of the unit tests failed after launching them by calling "make unit-tests && ctest --verbose -C Release" from the building directory (as it said in the manual), even when i didn't make any changes to the project at all

pavelkryukov, 2018-10-06:
> Clone repository with submodules: git clone ***--recursive*** https://github.com/MIPT-ILab/mipt-mips.git

I bet you missed 'recursive' while cloning.

> So should i clone that tracers brunch in order to build the project?

It is not a branch, it is a submodule. To fetch it in your case, type `git submodule update --init`

> Some of the unit tests failed after launching 

Some tests use traces from the submodule, so they should be fixed if you fetch and build them.

pavelkryukov, 2018-10-26:
@agrachiv99 I see you have some experience with Java. We have something to code in that language as well: https://github.com/MIPT-ILab/mipt-mips/issues/356

closed 2018-10-26 by pavelkryukov