MIPT-ILab/mipt-mips #564: Submit a patch to GNU Binutils to make the code C99-compliant.

pavelkryukov, 2018-09-16:
https://github.com/bminor/binutils-gdb/blob/75cbc781e371279f4403045be93b07fd8fe7fde5/sim/common/sim-core.h#L92

This code was introduced back in 1998, and C99 appeared next year, defining the `xor` as an 'alternative keyword' if `iso646.h` header is included. Moreover, C++98 defines it on the language level. As a result, the code is not buildable with C++ compiler or if `iso646.h` is included beforehand.

We have fixed it in our Binutils fork already, but the patch should be submitted to the master repository as well. Here is the instruction how to submit a patch: http://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blob;f=gdb/CONTRIBUTE;hb=HEAD

trexxet, 2018-09-16:
I think there is some reason why it's called `xor`, because near struct `_sim_core` has field `byte_xor`, not `xor`. Renaming it to `byte_xor` makes it compile, but may be logically incorrect. I'll try to look for usages and find if it can be named more correctly.

trexxet, 2018-09-28:
I've submitted patch to mailing list (https://sourceware.org/ml/gdb-patches/2018-09/msg00881.html). Let's wait for review.

trexxet, 2018-09-28:
Commited! [link](https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=commit;h=2283a21049cdefb87cdd0f11df367266e3e029d1)

pavelkryukov, 2018-09-28:
Congratulations!
(But it's better to set up an English name in your e-mail account)

closed 2018-09-28 by pavelkryukov

pavelkryukov, 2018-09-30:
I see an opportunity to make one more commit — the "ifdef __cplusplus" thing in simulator header.