MIPT-ILab/mipt-mips #589: Add ctor of MIPS instruction from string

pavelkryukov, 2018-09-29:
Currently the only way to construct a MIPS instruction is a decode it from 4 bytes. That's not convinient for tests (see #530), so the idea is to add a new constructor from stringed opcode:

```c++
MIPSInstr<MIPS32> instr( "sllv"); // Creates a SLLV instruction
```

agrachiv99, 2018-10-02:
What should I do in order to take this issue?

pavelkryukov, 2018-10-03:
I granted you write access to repository, and now you can put yourself as 'assignee' for that issue.
Please read our [contribution.md](
https://github.com/MIPT-ILab/mipt-mips/blob/master/contributing.md) to see our basic rules.

pavelkryukov, 2018-10-17:
@pooh64 Do you have any progress in these two weeks?

pooh64, 2018-10-18:
@pavelkryukov Now I'm exploring the structure of repository and c++ literature. I plan to finish this issue within a week.

commit 2a5e093d (2018-10-28): Add ctor of MIPS instruction from string

Fixes #589 
Additionally:
New utility method in class BaseMIPSInstr:
    MapType::iterator find_entry(MapType isaMap, std::string_view name);
New unit tests:
    TEST_CASE( "sllv") (#530)
    TEST_CASE( "MIPS32_instr: Divmult") ( using new ctor)

Fix types of supporting function.
Add get_v_imm and set_v_imm for unit tests.
Add sllv, mul, addi, teqi tests.
Fix nop bug for ctor, add nop test for new ctor.
closed 2018-10-28 by pavelkryukov