MIPT-ILab/mipt-mips #598: Rename 'complex pipeline' to 'non-unified pipeline'

pavelkryukov, 2018-10-03:
The word 'complex' has wrong similarity with complex numbers. We have to use more deliberate term for that options.

pavelkryukov, 2018-10-03:
Blocked by #301

ivababkin, 2018-10-11:
Sorry, can you explain this issue more exactly?

pavelkryukov, 2018-10-11:
I'm waiting for #301 to be completed to understand what to do.
But probably task would be so easy, so I'll do it myself.

ivababkin, 2018-10-15:
I'd like to do it by myself, but I dont have much time, I'll try to do it in 2-3 days. Sorry for waiting.--Отправлено из мобильной Яндекс.Почты11.10.2018, 19:12, "Pavel I.

pavelkryukov, 2018-10-15:
Basically the idea is to purge 'complex' from the code and documentation, and use term 'long' instead (long ALU).

ivababkin, 2018-10-15:
It's little bit uncomfortable to understand, what to do and just to complete those tasks, because we didn't work in this format. That's why I'm so slow.(--Отправлено из мобильной Яндекс.Почты15.10.2018, 22:05, "Pavel I.

pavelkryukov, 2018-10-15:
> because we didn't work in this format.

It's better to start earlier in a 'training' mode.

pavelkryukov, 2018-10-15:
> I'll try to do it in 2-3 days. 

Sorry to disappoint, but I did it in 10 minutes. My usual estimation is 1 point = 45 minutes of pure coding process.
When I opened the issue, it seemed to bit a little more complicated, but with Denis's explanation it become too easy to be a challenging task by any means.

pavelkryukov, 2018-10-15:
Instead, you may proceed with #643 — it requires the same C++ skills, but a little bit more work than running `sed -i 's/complex/long/'`

PR #647 (2018-10-15): Rename complex ALU to long ALU to avoid similarity with complex numbers
Fixes #598
commit f5f7ded1 (2018-10-15): Rename complex ALU to long ALU to avoid similarity with complex numbers (#647)

Fixes #598
closed 2018-10-15 by pavelkryukov