MIPT-ILab/mipt-mips #602: Enhance Trap class

pavelkryukov, 2018-10-04:
#601 introduces a Trap enumeration.

We need to extend it to contain a class, similarly to MIPSRegister:
* constructor
* dumping method
* conversion to/from index (if needed for GDB, Pavel — to define)

pooh64, 2019-02-28:
How should I implement any of these conversion?

Create macro list from MIPS traps enumeration (trap_types.h), separate it.
Create template classes for RISC-V and MIPS traps using macro.

Is it correct?

pavelkryukov, 2019-03-01:
First of all, I think it should be ISA-agnostic.
For implementation of GDB conversion, you may use a `switch/case`. We cannot copy `signals.def` as it is GPL, and we cannot guarantee that Binutils repository is cloned somewhere.

For RISC-V converter, the macro should be the best option.

pavelkryukov, 2019-03-27:
Any progress?

pooh64, 2019-03-28:
Just finished CS coursework. I plan to finish this issue this week.

pooh64, 2019-03-31:
I wrote a RISC-V-like enumeration for MIPS traps. Can I do something similar for GDB-exceptions without copying signals.def? Or I have to write switch/case for from_gdb method without any defines.

pavelkryukov, 2019-03-31:
No, no, that's not what we need, at all, completely.

We need a single Trap enumeration for all purposes, abstracted out of kernels, GDB, ISA etc.
For instance: unaligned address. In RISC-V convention, that is encoded with 4. In GNU convention, that is coded with 145.

We generate exception while executing instruction:

```c++
void executer_load_addr( Instr* instr)
{
// ..
  if (!aligned(instr->mem_addr))
    trap = Trap( UNALIGNED_ADDRESS);
}
```

If we need to intergrate RISC-V simulator to the RISC-V toolchain, we'll convert trap result to RISC-V format:

```c++
int execute_risc_v_simulation_step( Simulator* s)
{
    auto trap = s->step(); // returns UNALIGNED_ADDRESS
    retrun trap.to_riscv_format(); // returns 4
}
```

Same for GDB. We have the following code now:

https://github.com/MIPT-ILab/mipt-mips/blob/79120212ded48f95902a449c61a0e505b3e67e5c/simulator/export/gdb/gdb_interface.cpp#L36-L52

Instead, that function should use only Trap internal convertions:

```c++
using GDBTrap = std::pair<enum sim_stop, int>;
static GDBTrap translate_trap( Trap mipt_trap, int exit_code)
{
    if ( mipt_trap != HALT)
        return GDBTrap( sim_stopped, mipt_trap.to_gdb_format());
}
```

pooh64, 2019-03-31:
There are no enumeration for GDB-signals in mipt-mips. I've temporary created one. What is the best way to do this?

Should I move internal conversion and dump methods to another file?

There are no one-to-one correspondence between trap types (BREAKPOINT, EXPLICIT_TRAP and GDB_SIGNAL_TRAP). Can I resolve this in BREAKPOINT favor?

pavelkryukov, 2019-04-01:
> There are no enumeration for GDB-signals in mipt-mips. I've temporary created one. What is the best way to do this?

Any working way is good.

> Should I move internal conversion and dump methods to another file?

That should be in trap.cc

> There are no one-to-one correspondence between trap types (BREAKPOINT, EXPLICIT_TRAP and GDB_SIGNAL_TRAP). Can I resolve this in BREAKPOINT favor?

In the most reasonable way.

commit a094f508 (2019-04-18): Enhance Trap class (#947)

Fixes #602
closed 2019-04-18 by pavelkryukov