MIPT-ILab/mipt-mips #614: Implement "likely" branches

pavelkryukov, 2018-10-07:
Blocked by #91 

The idea of "likely" branch is to cause a "short" branch mispredict to fetch new instruction from branch target rather than from PC + 4.

TBD

BEANefiT, 2019-05-03:
Hello. If I get it right, we should always fetch and execute the first instruction (from a branch target), even if the branch is not taken. So that programmer who wants to write efficient and correct code should place there an instruction which won't do any harm in case of not taken branch. Am I right?

pavelkryukov, 2019-05-03:
>  I mean we don't flush instruction in case of misprediction.

No, how would that be possible?

The idea is following: if we fetch a branch and **there is no information from branch prediction**, the next fetched instruction is always PC + 4:

```
Branch F D E B W
PC+4     F D E *
PC+8       F D *
PC+12        F *
Target         F D E M W
```

For _likely_ branches, we should fetch Target as soon as possible. We cannot do it immediately since target is not decoded on F stage, so we have to use target mispredict flush to have the following pipe:

```
Branch  F D E B W
PC+4      F *
Target      F D E M W
```

closed 2019-05-05 by pavelkryukov