MIPT-ILab/mipt-mips #626: Implement Delayed Branches

pavelkryukov, 2018-10-09:
TBD

agrachiv99, 2018-11-28:
It is not clear, where exactly i should add this option for delay slots.

pavelkryukov, 2018-11-28:
I don't know either. I suggest to start with FuncSim.

pavelkryukov, 2018-11-28:
Usually constructor is preferrable, as you are able to look configuration in all class methods.

agrachiv99, 2018-11-28:
So I should add function, that replaces instructions in instruction list, when delayed slots option is turned on?

pavelkryukov, 2018-11-28:
First of all, I advise to get sure you have the test trace which is failing without delayed branches.

> that replaces instructions in instruction list, when delayed slots option is turned on?

What is the "instruction list"?

agrachiv99, 2018-11-29:
> What is the "instruction list"?

List of instructions that run method need to proceed. I mean, changing instructions with each other shouldn't be the work of simulator, should it? In my vision in constructor, order of the instructions should be changed to minimize the number of delay slots, as it would be done before by compiler. Is my vision proper?

agrachiv99, 2018-11-29:
> Build MIPS tests with delayed slots. Currently they are disable by .set noreorder directive in MIPS assembly files.

Sorry for stupid questions, but i can't find anything to understand what should i do here. There are no tests for delayed slots in *traces* directory and i don't understand what *set noreorder directive* is.

pavelkryukov, 2018-11-29:
GitHub has a nice feature called "search":
https://github.com/MIPT-ILab/mips-traces/search?q=set+noreorder&unscoped_q=set+noreorder

By default, MIPS compiler/assembler assumes the CPU (or simulator) supports branch delays and therefore it finds independent instructions to fill the delay slot. Since MIPT-MIPS does not support it, we had to disable them by setting that directive to assembler files. If this directive is removed, MIPS assembler will reorder instructions to fill the branch delay slot, and simulation will fail.

pavelkryukov, 2018-11-29:
More detailed explanation how compiler/assembler works:

https://stackoverflow.com/questions/3807480/weird-mips-assembler-behavior-with-jump-and-link-instruction

pavelkryukov, 2018-11-29:
Actually, I never investigated deeply how assembler works and what it broke in torture tests — I just disabled reordering and TT started to work,

Let's start with simpler test provided on SO;

```
.set noreorder
.section .text
.globl __start

__start:
   addi $a0, $0, 100
   addi $a1, $0, 200 
   jal test
   jr $zero # Required to halt simulation
test:
   add $v0, $a0, $a1
   jr $ra
```

If we build it with current flow:
https://github.com/MIPT-ILab/mips-traces/blob/0c8d51184a9918cdc80e68c6355deec07092fc5f/Makefile#L21-L27

we should get something like the following code (you can check it with `mips-linux-gnu-objdump  -EL -D -b test.out`, not sure about exact options), which is perfectly executed on MIPT-MIPS at the moment.

```
Disassembly of section .text:

00000000 <__start>:
   0:   20040064    addi    a0,zero,100
   4:   200500c8    addi    a1,zero,200
   8:   0c000004    jal 10 <test>
   c:   00000000    nop
   10:   00000008    jr  zero
   14:   00000000    nop

00000010 <test>:
  18:   00851020    add v0,a0,a1
  1c:   03e00008    jr  ra
  20:   00000000    nop
  24:   00000000    nop
```

We need to get something like SO author had:
```
Disassembly of section .text:

00000000 <__start>:
   0:   20040064    addi    a0,zero,100
   4:   0c000003    jal c <test>    <--- Why is jal coming before addi?
   8:   200500c8    addi    a1,zero,200
   c:   00000008    jr  zero
   10: 00000000    nop

0000000c <test>:
   14:   03e00008    jr  ra  <--- Why is jr coming before add?
   18:   00851020    add v0,a0,a1
    ...
```

agrachiv99, 2018-11-29:
> mips-linux-gnu-objdump -EL -D -b test.out

So in order to test it i need to use MIPS *binutils* or something else?

pavelkryukov, 2018-11-29:
> So in order to test it I need to use MIPS binutils or something else?

Yes, that's the only way to build ELF files from MIPS assembler.
You should have installed it to run simulation unit tests, right?

agrachiv99, 2018-11-30:
I guess I should make copy of all tests with *noreorder* because both **functional simulator** and **performance simulator** use same tests, and paste new path here:

`TEST_CASE( "Torture_Test: Func_Sim")
{
    CHECK_NOTHROW( get_simulator_with_test<MIPS32>( TEST_PATH "/tt.core.universal.out")->run_no_limit() );
    CHECK_NOTHROW( get_simulator_with_test<MIPS32>( TEST_PATH "/tt.core32.le.out")->run_no_limit() );
    CHECK_NOTHROW( get_simulator_with_test<MIPS64>( TEST_PATH "/tt.core64.le.out")->run_no_limit() );
}`

pavelkryukov, 2018-11-30:
If you want to run these tests with `ctest`, yes, but at the beginning you may run them just from shell with `mipt-mips` command.

agrachiv99, 2018-11-30:
That is how disassembly looks like: 
```
004000d0 <__start>:
  4000d0:	20040064 	addi	a0,zero,100
  4000d4:	0c100039 	jal	4000e4 <test>
  4000d8:	200500c8 	addi	a1,zero,200
  4000dc:	00000008 	jr	zero
  4000e0:	00000000 	nop

004000e4 <test>:
  4000e4:	03e00008 	jr	ra
  4000e8:	00851020 	add	v0,a0,a1
  4000ec:	00000000 	nop
```

There are 2 instructions after the jump, and these 2 are always filled with nops or independent instructions, so it should work without any pipeline flushing or whatever, am I right?

pavelkryukov, 2018-11-30:
We do not touch pipeline at the moment, we are talking only about functional simulation (single-cycle implementation, func_sim.cpp).

> simulation should work without any pipeline flushing or whatever, am I right?

Please run functional simulation at your own and see what it does.

agrachiv99, 2018-11-30:
I've just understood that functional simulator is single-cycled, but then it can't face any branches hazards. So what does implementing delayed branches in single-cycled simulator stands for?

pavelkryukov, 2018-11-30:
> Please run functional simulation at your own and see what it does.

Please, do this and share what instructions are executed by default (`./mipt-mips -b test.out -d -f`).

agrachiv99, 2018-11-30:
I see the problem. So as soon as simulator meets branch condition it should execute 2 more instructions before jumping?

pavelkryukov, 2018-11-30:
> So as soon as simulator meets branch condition it should execute 2 more instructions before jumping?

Right, but we should optionally select 0, 1 or 2.
That's the problem with delayed branches method Oleg explained on the lecture — whereas it works well in pipelined microarchitecture, it still should be supported in other implementations, like single-cycle or more complicated.

agrachiv99, 2018-11-30:
> That's the problem with delayed branches method Oleg explained on the lecture — whereas it works well in pipelined microarchitecture, it still should be supported in other implementations, like single-cycle or more complicated.

Why do we need to optionally select 0, 1 or 2 if compiler always put 2 instructions after the branch condition

pavelkryukov, 2018-11-30:
> compiler always put 2 instructions after the branch condition

So far I don't see 2 instructions case, I see always 1 — please explain if I'm wrong

> Why do we need to optionally select 0, 1 or 2?

1. RISC-V has no branch delay option, thus we'll always choose 0 option
1. MARS and SPIM has an option to execute without branch delay — we need to have it as well
1. If compiler is instructed not to reorder instructions, we must have 0 (as we have at the moment).

pavelkryukov, 2018-11-30:
And, finally, we won't have a branch delay slot in PerfSim for some time, so we have to have no slots to ensure both simulators behave similarly.

agrachiv99, 2018-11-30:
This delayed option slots should be one of the command line parameters ideally, or i should just add it into constructor parameter as you said?

pavelkryukov, 2018-11-30:
Do as best as you can, we'll refactor that later.

agrachiv99, 2018-12-02:
Is alu.h related only to functional simulator?

pavelkryukov, 2018-12-20:
Committing 1 point as something like a solution was proposed

pavelkryukov, 2019-02-27:
Closing as duplicate of #737

closed 2019-02-27 by pavelkryukov